OrcaSlicer/OrcaSlicer #425: Kubuntu 22.04 missing libwebkit2gtk-4.0.so.37 when running AppImage

Guswilly, 2023-03-04:
**Describe the bug**
Steps taken:
Downloaded zip file,  extracted, properties changed to executable, then
Run in Console returns:
/tmp/.mount_BambuSdizE6Q/bin/bambu-studio: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

Ran updates, and I have tried with the same error:
sudo apt install libfuse2
Unfortunately Web based content is overloaded with the above fix making it hard to find a fix.

**Desktop (please complete the following information):**
 - OS: Kubuntu 22.04

cereal7802, 2023-03-09:
Did you install libwebkit2gtk-4.0? I had to install webkit2gtk4.0 package on Fedora to get rid of that error.

Fedora: dnf install webkit2gtk4.0
(K)Ubuntu: should be apt install libwebkit2gtk-4.0

It might be a different package name as searching online it seems like it has libwebkit2gtk-4.0-37, but it might be different for your specific version of kubuntu.

royw, 2023-03-10:
Same issue on manjaro stable.
` pacman -Ss libwebkit2gtk                                                                                                                                                                                                          
extra/webkit2gtk 2.38.4-1
    Web content engine for GTK
extra/webkit2gtk-4.1 2.38.4-1
    Web content engine for GTK
extra/webkit2gtk-5.0 2.38.4-1
    Web content engine for GTK
`

Guswilly, 2023-03-13:
Thanks for the help.  I tried this all again and found that I had a typo in the library name, a simple . versus a -   (libwebkit2gth-4.0.so-37 is the correct name).  

Now it works.

khumarahn, 2023-06-26:
Same here on gentoo. I have `net-libs/webkit-gtk-2.40.1-r410` installed. I don't undestand how to fix this

Krutonium, 2023-07-03:
This *really* should be packed into the AppImage

lordfiSh, 2023-09-10:
Same Problem on the Steam Deck (Arch Linux)
`/tmp/.mount_OrcaSlPfBnnd/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory`

Cypher1, 2023-12-07:
For me the following packages were missing:

```
webkit2gtk4.0-2.40.0-2.fc38.x86_64
mesa-libOSMesa.x86_64
```
i.e.
```
sudo dnf install webkit2gtk4.0-2.40.0-2.fc38.x86_64 mesa-libOSMesa.x86_64
```

I also suspect the following may be helpful for the live preview, but haven't gotten it working.
```
sudo dnf install gstreamer1-plugins-bad-freeworld.x86_64 gstreamer1-svt-av1.x86_64 gstreamer1-plugins-bad-free-extras-1.22.5-1.fc38.x86_64
```

magealexstra, 2024-01-15:
just fyi i'm running kubuntu 23.10 and to install i had to search and found it listed as libwebkit2gtk-4.0-37 in apt-cache.

installing allowed the app to launch. 

```
sudo apt install libwebkit2gtk-4.0-37
```

DupiDachs, 2024-03-13:
I am under Archlinux here and the package webkit2gtk4.0 is not provided here. But there is webkit2gtk4.1.

The workaround is to install version 4.1 and then create some symbolic links.

`
me@AsusLaptop: $ ldconfig -p | grep webkit
	libwebkit2gtk-4.1.so.0 (libc6,x86-64) => /usr/lib/libwebkit2gtk-4.1.so.0
	libwebkit2gtk-4.1.so (libc6,x86-64) => /usr/lib/libwebkit2gtk-4.1.so
`
and hence (you need to be in the folder of your AppImage),
`
me@AsusLaptop: $ ln -s /usr/lib/libwebkit2gtk-4.1.so.0 libwebkit2gtk-4.0.so.37
`
and also
`
me@AsusLaptop: $ ldconfig -p | grep libjava
	libjavascriptcoregtk-4.1.so.0 (libc6,x86-64) => /usr/lib/libjavascriptcoregtk-4.1.so.0
	libjavascriptcoregtk-4.1.so (libc6,x86-64) => /usr/lib/libjavascriptcoregtk-4.1.so
`
and hence (you need to be in the folder of your AppImage),
`
me@AsusLaptop: $ ln -s /usr/lib/libjavascriptcoregtk-4.1.so.0 libjavascriptcoregtk-4.0.so.18
`

Does the job for me. HF!

czinehuba, 2024-04-29:
Slightly different directories, but it in Ubuntu 24.04 did the trick. 

sudo ln -s /lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0 /lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
sudo ln -s /lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0 /lib/x86_64-linux-gnu/libjavascriptcoregtk-4.0.so.18

SebiPanther, 2024-05-01:
That worked for me. I also installed the following packages (maybe not all of them needed):
`sudo apt-get install libwebkit2gtk-4.1-dev libwebkit2gtk-4.1-0 language-pack-en language-pack-en-base language-pack-gnome-en language-pack-gnome-en-base language-pack-kde-en`

SebiPanther, 2024-05-01:
> Slightly different directories, but it in Ubuntu 24.04 did the trick as well. cheers
> 
> sudo ln -s /lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0 /lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37 sudo ln -s /lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0 /lib/x86_64-linux-gnu/libjavascriptcoregtk-4.0.so.18

That worked for me. I also installed the following packages (maybe not all of them needed):
`sudo apt-get install libwebkit2gtk-4.1-dev libwebkit2gtk-4.1-0 language-pack-en language-pack-en-base language-pack-gnome-en language-pack-gnome-en-base language-pack-kde-en`

steve7233, 2024-05-25:
> Slightly different directories, but it in Ubuntu 24.04 did the trick as well. cheers
>
> sudo ln -s /lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0 /lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37 sudo ln -s [...]

Did not work for me. Kubuntu 24.04

steve7233, 2024-05-25:
> Slightly different directories, but it in Ubuntu 24.04 did the trick as well. cheers
>
> sudo ln -s /lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0 /lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37 sudo ln -s [...]

Did not work for me. Kubuntu 24.04

> Did you install libwebkit2gtk-4.0? I had to install webkit2gtk4.0 package on Fedora to get rid of that error.
>
> Fedora: dnf install webkit2gtk4.0 (K)Ubuntu: should be apt install libwebkit2gtk-4.0 [...]

Package not found. Kubuntu 24.04.

mikey54, 2024-06-03:
> That worked for me. I also installed the following packages (maybe not all of them needed): `sudo apt-get install libwebkit2gtk-4.1-dev libwebkit2gtk-4.1-0 language-pack-en language-pack-en-base [...]

I did this and now I have the error:

"authentication failed due to unexpected error"

Kubuntu 24.04

BadPixel89, 2024-08-15:
This is affecting Mint as well. I resolved by doing the following:

Create the links as outlined in a previous post:
`sudo ln -sf /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37`
`sudo ln -sf /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.0.so.18`

Move the appimage to a new folder:
`/home/BadPixel89/tools/OrcaSlicer/OrcaSlicer_Linux_V2.1.1.AppImage`

cd to the directory containing the app image and run the it from terminal with the following:
`WEBKIT_DISABLE_COMPOSITING_MODE=1 ./OrcaSlicer_Linux_V2.1.1.AppImageD`

This allows you to run through the setup, choose a printer / filament profiles etc

Then I followed [this guide](https://ekiwi-blog.de/en/61930/linux-mint-add-entries-to-the-start-menu/) to add the appimage as a shortcut to the start menu. Launching from here everything seems to work but I can't add new presets. I suppose if I want to do that I will need to run the appimage from terminal or figure out how to add the parameter to the start menu item.

Seems like the only bit that's affected is the initial setup wizard and the profile picker.

ExperimentalCyborg, 2024-09-02:
> Did you install libwebkit2gtk-4.0? I had to install webkit2gtk4.0 package on Fedora to get rid of that error.
>
> Fedora: dnf install webkit2gtk4.0 (K)Ubuntu: should be apt install libwebkit2gtk-4.0 [...]

Did not work for me on fedora 40, while it does start up after installing that package the welcome screen is blank. :(

BadPixel89, 2024-09-03:
> While it does start up after installing that package the welcome screen is blank. :(

Did you run it with the argument like this, I had the same result when not using the argument:

`WEBKIT_DISABLE_COMPOSITING_MODE=1 ./OrcaSlicer_Linux_V2.1.1.AppImageD`

ExperimentalCyborg, 2024-09-04:
> > While it does start up after installing that package the welcome screen is blank. :(
> 
> Did you run it with the argument like this, I had the same result when not using the argument on Mint:
> 
> `WEBKIT_DISABLE_COMPOSITING_MODE=1 ./OrcaSlicer_Linux_V2.1.1.AppImageD`

Oh crap i missed that detail. Thank you so much!

xpcone, 2024-09-11:
WEBKIT_DISABLE_COMPOSITING_MODE=1  did not help, webkit 4.0 is too old

ExperimentalCyborg, 2024-09-11:
Yeah i happily thanked badpixel after the gui showed up but the 3D render in the workspace still doesn't show up :joy: Until then i'll be using a laptop running mint 22 for slicing... :smiling_face_with_tear:

olimexsmart, 2024-11-24:
Same error here: `/tmp/.mount_OrcaSlU9Wgv2/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.1.so.0: cannot open shared object file: No such file or directory`

Debian 12 with KDE Plasma

TeamFahQ, 2024-12-25:
> EDIT: I manage to solve downloading this Flatpack: https://github.com/anarsoul/io.github.softfever.OrcaSlicer/releases/tag/v2.2.0-1

Worked for me in Linux Mint 22. Unfortunately, there is no "dark" mode theme on this version.,

Beep6581, 2025-03-06:
This seems to be the oldest open issue regarding `libwebkit2gtk-4.0.so.37` and the AppImage. Someone really should close the multitude of open duplicates:
https://github.com/SoftFever/OrcaSlicer/issues?q=is%3Aissue%20state%3Aopen%20libwebkit2gtk-4.0.so.37

I accidentally posted in this closed issue:
https://github.com/SoftFever/OrcaSlicer/issues/4616#issuecomment-2704402878

Beep6581, 2025-06-09:
.

olimexsmart, 2025-07-22:
Fair enough, this can be closed IMHO

closed 2025-10-29 by github-actions[bot]