OrcaSlicer/OrcaSlicer #427: BambuStudio-SoftFever will not open after latest macOS update

radaiko, 2023-03-05:
After the latest macOS update BambuStudio-SoftFever is not allowed to start.

[image]

I tried to remove it completely from the Application folder an redownload it and run 
`xattr -d com.apple.quarantine /Applications/BambuStudio-SoftFever.app` 
but still the same message appears
It was running before.

I'm using an MacBook Air M1 with macOS 13.2.1

Kerber0ss, 2023-03-05:
Same problem

hliebscher, 2023-03-05:
xattr -dr com.apple.quarantine /Applications/BambuStudio-SoftFever.app

with -dr

Kerber0ss, 2023-03-05:
> xattr -dr com.apple.quarantine /Applications/BambuStudio-SoftFever.app
> 
> with -dr

still not working

jzelez, 2023-03-05:
If it helps I got this in the log:
`/Applications/BambuStudio-SoftFever.app/Contents/MacOS/BambuStudio not valid: Error Domain=AppleMobileFileIntegrityError Code=-420 "The signature on the file is invalid" UserInfo={NSURL=file:///Applications/BambuStudio-SoftFever.app/Contents/MacOS/BambuStudio, NSLocalizedDescription=The signature on the file is invalid}`

`mac_vnode_check_signature: /Applications/BambuStudio-SoftFever.app/Contents/MacOS/BambuStudio: code signature validation failed fatally: When validating /Applications/BambuStudio-SoftFever.app/Contents/MacOS/BambuStudio:
  The code contains a Team ID, but validating its signature failed.`

`proc 85436: load code signature error 4 for file "BambuStudio"`

`ASP: Security policy would not allow process: 85436, /Applications/BambuStudio-SoftFever.app/Contents/MacOS/BambuStudio`

goldjunge91, 2023-03-05:
> 

try this 
make a folder in applications name it Slicer,
sudo -s 
xattr -dr com.apple.quarantine /Applications/Slicer/

for me on my MacBook Air M1 its full working after this and you have to download the [BambuStudio-SoftFever_V1.4.5_Mac_x86_64.zip](https://github.com/SoftFever/BambuStudio-SoftFever/releases/download/v1.4.5/BambuStudio-SoftFever_V1.4.5_Mac_x86_64.zip)

Highlander-bdj, 2023-03-05:
The Slicers trick only works with th ex86/64 fiel.   I can't get the ARM version to run.

markuswarlich, 2023-03-06:
Me too. No chance with the M1

ab3dgeek, 2023-03-06:
same issue came up just out of the blue yesterday for me too.
(there might have been some security updates to the os, I'm not sure)

I solved it by deleting the arm version, copying the x86_64 in its place and I then run the xattr command in terminal and it works again.

goldjunge91, 2023-03-06:
> The Slicers folder trick only works with the x86/64 file. I can't get the ARM version to run.

oh okay i was thinking the x86/64 is the arm version. i was not sure what the right one is i try both and one of the files work.

markuswarlich, 2023-03-06:
Thanks for the tip. The X86/64 version works for me.

esc247, 2023-03-06:
none of the versions work for me even with the slicer trick. Do you need to use a specific browser to download the file?

Kerber0ss, 2023-03-06:
> It looks like there are some changes in the latest OSX.
> Good news: I have the Apple Develop ID ready and should be able to tease a signed build pretty soon.

I don’t update to 13.2.1 at m1 and have that bug too, all works until this weekend. Today I update, still not work.

IanAlbinson, 2023-03-06:
Confirmed here, too. The x86/64 version works with the "xattr -d com.apple.quarantine" Terminal command. The arm version does not.

Highlander-bdj, 2023-03-08:
Any chance on a quick fix?  The X86 version is SLOW on an M1 Mac. :)

aikoradlingmayr, 2023-03-08:
Please don‘t stress him. He is doing this awesome work in his free time and is getting no money for that. 

I have made so many negative experiences in other open source project where the owner gets insulted for not working fast enough (look at the superslicer repo) so please wait patient. He already respond in this issue so he will do as much as he can.

bistory, 2023-03-08:
> Please don‘t stress him. He is doing this awesome work in his free time and is getting no money for that.
>
> I have made so many negative experiences in other open source project where the owner gets [...]

I second this, SoftFever made an awesome work by sharing this software ! If you want to fix this bug faster, don't hesitate : it's an open-source project, this means you can grab sources, fix the bug and make a pull request to help the author of the repo. Otherwise, relax, be patient, he has a work and a life outside Github

olsonjj, 2023-03-08:
> I second this, SoftFever made an awesome work by sharing this software ! If you want to fix this bug faster, don't hesitate : it's an open-source project, this means you can grab sources, fix the [...]

If this was a matter of fixing the source, I'd be glad to do it.  The issue here is its an unsigned application and to run it before even this, required to bypass security policy.

I'd be happy to contribute towards the yearly fee of $99 for the author Apple dev license. In fact, I'd love to see the slicer added to the app store, even adding a $4.99 fee to buy it. I'd gladly pay.

Highlander-bdj, 2023-03-08:
> If this was a matter of fixing the source, I'd be glad to do it. The issue here is its an unsigned application and to run it before even this, required to bypass security policy.
>
> I'd be happy to [...]

I'd be happy to as well, or setup a Patreon and I'll be in line to sign up to support the author.

rlabusiness, 2023-03-08:
> I'd be happy to contribute towards the yearly fee of $99 for the author Apple dev license. In fact, I'd love to see the slicer added to the app store, even adding a $4.99 fee to buy it. I'd gladly pay.

Although I'm pretty bought-in to Apple's ecosystem, I'd prefer this not be distributed solely through the App Store since that gives Apple the opportunity to shut things down or change policy to invalidate updates. That's happened to many beloved apps over the years, and I'm tired of it.

In my opinion, all apps you previously installed should always be available in perpetuity, even if they are later removed from the App Store, but that's not how Apple works, unfortunately. I value having SoftFever's version of Bambu Studio outside of that delivery mechanism.

asoderberg, 2023-03-08:
> Although I'm pretty bought-in to Apple's ecosystem, I'd prefer this not be distributed solely through the App Store since that gives Apple the opportunity to shut things down or change policy to [...]

Don't equate getting an Apple Developer license with publishing apps through the Apple App Store. While an Apple Dev license is required to publish to the App Store. Developers are NOT required to publish via the App Store. Developers can publish their Apps (digitally signed via the Dev license) in any way they want, including here on GitHub.

Manfredo1952, 2023-03-10:
Someone with an M1 MAC mentioned that he simply compiled the app himself. He cloned the project the project ionto his harddisk and used the terminal to start the compile session with the .sh command contained in the root of the project build_release_macos.sh

radaiko, 2023-03-10:
Thanks, for the quick fix.

Switch-Witch, 2023-03-10:
You saved my day/week/year/live.... You are my hero!! Thx so much!!

roamingthings, 2023-03-11:
Thank you for being transparent about that 👍

SoftFever, 2023-03-15:
Sorry for all the inconvenience. I was unaware we need a Github account to download builds from the Actions page.
I've re-uploaded the un-signed version to the 1.4.5 release page: https://github.com/SoftFever/OrcaSlicer/releases/download/v1.4.5/BambuStudio-SoftFever_V1.4.5_Mac_arm64_nosign.zip

closed 2023-11-26 by github-actions[bot]