OrcaSlicer/OrcaSlicer #428: Bambu Studio Info. Too large layer height. Reset to 0.500

olivogt, 2023-03-05:
Hi - is the above intentional? At the moment it seems I cannot set layer heights larger than 0.5mm.
This would mean no support for 0.8mm nozzles and above - right?

Thanks,
Oliver

SoftFever, 2023-03-05:
this value limits layer height, you need to change it 
[image]

olivogt, 2023-03-05:
I was just about to say I found it too ... ;-)

Thanks,
Oliver

closed 2023-03-06 by SoftFever

domik82, 2023-03-26:
I honestly think it should be mentioned somewhere in the popup box that it can be changed and how to find it. 

I'm trying to print this Fast-Printing Modular Drawer System (Vase Mode) - https://www.printables.com/model/139570-fast-printing-modular-drawer-system-vase-mode

And it suggests that:

> I use a 0.8mm line width for higher strength (a 0.6mm line width should also give good results, but will be less strong), 0.32 layer height, and 25-30mm/s wall speed. 
> 
> I use a 0.4mm nozzle for all parts. 
> 
> No supports required for any parts.

It was quite difficult to figure out where I should change something to try to print with the same settings.