OrcaSlicer/OrcaSlicer #431: Implement BugFix for unexpected seam in vase mode

dojo90, 2023-03-06:
I currently have the problem that a seam can be seen in vase mode. This is probably a problem that is in the original source code.

`degroof` has probably managed to write a small hotfix for it in java. Would it be possible to implement this as an experimental fix (optionally with a checkbox)?
Should this feature be implemented in BambuStudio, especially in the SoftFever Fork, this will again be an enormous boost for our community.

https://github.com/prusa3d/PrusaSlicer/issues/2841#issuecomment-1267667080

### screenshots:

[image]

### sample stl file

https://anonfiles.com/L1y8xfc1z8/peel_stl

Pmant, 2023-03-06:
This is a post processing script and would not work during slicing. You probably can add it as a post processing script, which is supported by the SoftFever fork. 
There is always a risk when using such scripts to allter your gcode as they can easily do so in unintended ways. The author of the script describes it as alpha and that it does things like removing color changes.

npoulos, 2023-03-10:
Where can you find the option to add post processing scripts in softfever? Didn’t know this was available!

Pmant, 2023-03-10:
Under process settings -> others. Personally I have not used it so far.

closed 2023-11-26 by github-actions[bot]