OrcaSlicer/OrcaSlicer #432: Shrinkage Compensation

DanielGroot, 2023-03-06:
**Is your feature request related to a problem? Please describe.**
My filaments are kinda wonky so i need some shrinkage compensation for some filaments. In superslicer there's an option for that, but I'm missing it in BambuStudio. Therer is a x-y countour compensation but that doesnt just scale up the model by a given procent

**Describe the solution you'd like**
I'd like a feature where you type in a percentage that you calculate by printing a small cube and see how much smaller it's than you requested. And the slicer will compensate for that by scaling up that model by that x procent

SoftFever, 2023-03-06:
Hi,

We already have a ticket for this FR #360 
This feature has been implemented in the dev branch, and will be available in the next release.
If you want to try now, download the latest nightly build here: https://github.com/SoftFever/BambuStudio-SoftFever/actions/workflows/build_win.yml.

SoftFever, 2023-03-06:
Feel free to comment on that ticket.
I'm closing this one as it's duplicated.

closed 2023-03-06 by SoftFever