OrcaSlicer/OrcaSlicer #439: Add retraction calibration tool

erickdredd, 2023-03-07:
The new calibration tools have been fantastic, however retraction calibration is sorely missed now that just about everything else is so easy and convenient. I would love to see something like the towers generated by [this tool](http://retractioncalibration.com/) built into the slicer, as it's one of the best and quickest ways to tune retraction that I'm personally aware of.

ab3dgeek, 2023-03-08:
yes that one looks like a good tool, it sure would be a nice and much needed addition to the already nice selection of calibration tools.

adamfilip, 2023-03-12:
Please also include the ability to tune Detraction/extra restart as well.

SoftFever, 2023-05-21:
Added in 1.6.3 beta

closed 2023-05-21 by SoftFever

nilbest, 2024-01-03:
I know this is a closed Feature request, but I'm unsure if this is rolled out in 1.8.1.
I'm missing the feature for setting up the retraction speed. A toggle implemented in the other calibration tools would be great.
For example, setting up TPU with a Bowden extruder is right now a pain in the ass because of that.
I still have to set it up in Curra first or use this page https://teachingtechyt.github.io/calibration.html#retraction .
[image]
An in-slicer integrated function would be great. Maybe a mash of the normal retraction distance is changeable via a selection like in the temp tap.
[image]
[image]
[image]