OrcaSlicer/OrcaSlicer #442: Start-Up  Error `index.html:1:9: ReferenceError: Can't find variable: __wxOut0`

norpol, 2023-03-07:
**Describe the bug**

Starting up BambuStudio-SoftFever sometimes shows up an error like this one, although everything appears to be functional after closing the dialog.

 > `Error running JavaScript: file:///nix/store/m0aq9bm6rn2ym3r66z21cp57prlvb8v6-bambu-studio-softfever-1.4.5/share/BambuStudio/web/homepage/index.html:1:9: ReferenceError: Can't find variable: __wxOut0`
or

> `"Error running Javascript: file:///tmp/.mount_BambuSKzd8iH/resources/web/homepage/index.html?lang=en_GB:1:9: ReferenceError: Can't find variable: __wxOut0"`

if you are using the AppImage https://github.com/SoftFever/BambuStudio-SoftFever/pull/86

**To Reproduce**
Open bambu-studio several times, 

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Linux
 - Version: v1.4.5

kmarty, 2023-03-22:
Just FYI, the problem persist in OrcaSlicer v1.5.0.

fleutot, 2023-03-24:
Confirmed, and seeing OP's screenshot is interesting: they're apparently in some kind of esoteric window manager, probably one that doesn't do reparenting. 

I'm seeing the same issue, and I am also using a non-reparenting WM (xmonad). How about you @kmarty ?

kmarty, 2023-03-24:
I have "standard" Gnome installation in Debian11  (i.e. Gnome on Wayland)

shizhx, 2023-03-29:
Same problem on Manjaro Linux
```bash
Operating System: Manjaro Linux
KDE Plasma Version: 5.26.5
KDE Frameworks Version: 5.103.0
Qt Version: 5.15.8
Kernel Version: 5.15.94-1-MANJARO (64-bit)
Graphics Platform: X11
Processors: 8 × 11th Gen Intel® Core™ i5-11320H @ 3.20GHz
Memory: 15.4 GiB of RAM
Graphics Processor: Mesa Intel® Xe Graphics
Manufacturer: realme
Product Name: CloudProXXXX
System Version: MP
```
[image]

KrzysztofKam, 2023-04-09:
Same error here. In both: OrcaSlicer_ubu64.AppImage version 1.5 and version 1.6.1. Tested on two systems: Kubuntu 23.04 beta and Kubuntu 22.04 LTS.

ChaosBlades, 2023-04-17:
Duplicate of https://github.com/SoftFever/OrcaSlicer/issues/219

norpol, 2023-04-28:
This appears to be an upstream bug and not OrcaSlicer only issue.

https://github.com/bambulab/BambuStudio/issues/12#issuecomment-1511424730

Pastitas, 2023-05-08:
There seems to be some news, stays unfixed on linux by the looks of it

brychanrobot, 2023-07-26:
It looks like the upstream bug got fixed in 1.7. When can we expect 1.7 to make its way into OrcaSlicer?

norpol, 2023-11-17:
https://github.com/SoftFever/OrcaSlicer/issues/219#issuecomment-1712811747

closed 2023-11-17 by norpol