OrcaSlicer/OrcaSlicer #443: First layer travel speed

npoulos, 2023-03-08:
**Is your feature request related to a problem? Please describe.**
Is it possible to be able to add a travel speed for the first layer? 
I try to print the first layer really slow but the travels kind of mess that up.

Ibot-11, 2023-03-08:
Just lower your accelerations. This is often the reason for bad first layers. I use 1000mm/s²
Also use a Z-hop of ~0,2mm
A slow travel speed on the first layer can cause oozing and mess up your first layer even more.

rovili, 2023-03-08:
I would actually request it a bit differently: 
Split the Travel & First-Layer acceleration -> you can use slow acceleration for printing the first-layer but fast travel between printing points.
This wouldn't influence the adhesion but can boost printing time a lot.

npoulos, 2023-03-08:
I print hundreds of very tiny miniatures per place. My first layer speed is 8mm/s in order to make sure it has perfect adhesion. I don’t want extremely slow travels just slower than regular layer travels. Travel acceleration I already have low as it improves seams a lot. 

Not really looking for a solution other than a bottom layer travel speed setting. If @SoftFever doesnt think it’s worth it all good but it would help me out a lot.

Ibot-11, 2023-03-09:
Why do you have slow travel speeds on your first layer? It’s the same as for every other layer. First layer travel acceleration can be useful, but to be honest I don’t see many benefits. Simply increase your first layer accel to ~1000-2000mm/s² and your problem will be fixed.

rovili, 2023-03-09:
> Why do you have slow travel speeds on your first layer? It’s the same as for every other layer. First layer travel acceleration can be useful, but to be honest I don’t see many benefits. Simply increase your first layer accel to ~1000-2000mm/s² and your problem will be fixed.
> 
> Your slow toolhead move after your primeline is a macro / startgcode issue. It simply uses the last provided feedrate.

It is not - first-layer acceleration is applied to travel on the first layer. That's simply not correct.

Ibot-11, 2023-03-09:
Sure, the first layer acceleration is applied to the travel moves. That's why I recommend just increasing it a bit. As far as I have seen the default is 500mm/s² on most default profiles. This is way too slow in my option. 1500mm/s² is a good value for me. Sure I don’t reach my 500mm/s travel on small moves on the first layer. But it’s still fast enough. It’s just the first layer. The time saving for high acceleration travel moves on the first layer will be just a few seconds.

dckhb, 2023-03-09:
> Sure, the first layer acceleration is applied to the travel moves. That's why I recommend just increasing it a bit. As far as I have seen the default is 500mm/s² on most default profiles. This is way too slow in my option. 1500mm/s² is a good value for me. Sure I don’t reach my 500mm/s travel on small moves on the first layer. But it’s still fast enough. It’s just the first layer. The time saving for high acceleration travel moves on the first layer will be just a few seconds.
> 
> Don’t get me wrong. New settings are great. I just would like to avoid that the settings look like in Cura. Way too much stuff that nobody really needs / tunes for their printer.
> 
> It's just my opinion. In the end SoftFever will decide if this will be added or not

mate, just F.off and let people ask for what they need. as you said this isn't your call anyway. if they need that feature who are you to come and just say 'set your acceleration this speed that and your problems are solved' what an ignorant idiot...
and then you have an issue opened cause you don't like the automatic placement function, ffs grow up.

Ibot-11, 2023-03-09:
Sorry, maybe I should have written the message a bit differently. I just wanted to discuss this topic and also share my experience and opinion on this. I hadn't expected that this would escalate like this.
But please stay respectful. Words like “mate, just F.off” are really not needed.
I’ll also try my best to write this a bit more “friendly” next time.

npoulos, 2023-03-09:
While I’d say discussion is encouraged, ibot you’re coming across extremely dismissive. 

My specific use case is that I print tiny pieces with a .2 nozzle and 2 walls wide (no infill). First layer is usually 0.08mm or less. 

Basically perfect conditions for things to mess up. After the first (or first few) layers everything is good to up the speeds and travels but on the first layer travel (along with travel acceleration) needs to be slow. 

My first layer is printed super slow at 8mm/s so the speed up from such a slow speed can cause issues. 

These features are all available in Cura so there’s obviously other reasons you’d want to control this but this is my use case. 

I’m not looking to discuss or problem solve this issue with other methods as I’ve tested them all already and they don’t produce the results needed which is why I’m requesting the feature.

SoftFever, 2023-03-13:
Implemented.
Will be available in 1.5 version

npoulos, 2023-03-13:
genius!

closed 2023-03-17 by SoftFever

RudiKleinSA, 2023-04-17:
Hi, 

I am enjoying ORCA very much - Like to save PROJECTS with PROJECT SETTINGS and the MANY BOARDS.

I only print ABS, and warping is a problem for me - especially square corners.

I am on Version 1.6 and see improvements made above - thanks.

However,  I usually like to print about x10 slow layers, which gives me about 2mm thick bottom, and then go to the normal print speed.

I would like a setting to set the number of slow layers.

Slow initial layers guarantee my long prints ahead. LOL. Saving me a lot of time with guaranteed prints.

I currently do this manually in Mainsail.

Alternatively if someone could give me a start G Code for my printer to always do this with ABS

Thanks

vgdh, 2025-09-28:
I wrote a fix for a low first-layer acceleration 
https://github.com/vgdh/OrcaSlicer-first-layer-travel-acceleration-fix