OrcaSlicer/OrcaSlicer #444: Allow only full 90° rotations when using “Auto rotate for arrangement”

Ibot-11, 2023-03-08:
**Is your feature request related to a problem? Please describe.**
When I add a lot of parts to the build plate the arrangement tool doesn’t arrange them well. When you enable “Auto rotate for arrangement” the results are better.
Unfortunately some parts are rotated 45° which causes a non matching infill pattern compared to the other parts.

**Describe the solution you'd like**
A new checkbox under the “Auto rotate for arrangement” checkbox where you can only allow full 90° rotations.

**Describe alternatives you've considered**
Arrange all 45° rotated parts manually to match the others (90°)

**Additional context**
[image]

Ibot-11, 2023-03-09:
Just another picture to visualize my "problem" with the 45° rotation a bit better. Maybe it’s a bit hard to understand the above text without a picture.
[image]

Maybe I’m just really picky with my first layer and print quality in general. But if the first layer pattern isn’t 45° like the others it looks a bit weird to me.

kylek29, 2023-03-10:
Yeah, that would be a useful feature ... sometimes with more complex models, the program just decides to put it at a severe angle.

To add on to that, I think having a modifier key (e.g. holding Shift) for when you're manually rotating to force it to snap to whole numbers (e.g. when using the mouse to rotate, it'd snap to 90, 91, 92, 93, and so fourth).

mrwilby, 2023-05-13:
Perhaps tangentially related, but I'd also like the ability to always rotate all parts at 45 degrees:

https://github.com/AndrewEllis93/Ellis-SuperSlicer-Profiles#45-degree-profile-vs-standard-profile

julie777, 2023-06-05:
I accidentally discovered another reason to be able to restrict auto arrange to 90 degree rotations.

I was trying to position a problematic model on the plate in a way to get a certain camera view. That way I could watch progress and stop if the issue because visible. I did not fail and I quit watching. When I pulled the print of the printer it was unusable because I had chosen a terrible angle and the bottom layer and infill ruined the chamfer on the outside.

[image]

closed 2023-11-26 by github-actions[bot]

mtczekajlo, 2024-09-12:
I'm about to print parts for V-Minion and I thought that would be a great feature indeed. Can this be considered again? 🙏