OrcaSlicer/OrcaSlicer #451: Creality Sonic Pad (klipper) not showing up in Device tab

rlabusiness, 2023-03-08:
Due to how the Creality Sonic Pad web interface is configured, it doesn't load in SoftFever Device tab when you configure the slicer to send a print to the network port assigned to your printer (e.g. 7127).

To fix this compatibility issue and give users a bit more flexibility over their web configuration, please add a field to override the Device tab's URL in case it is different from the Octoprint/moonraker API URL.

### I've provided more detail about this in Discussion https://github.com/SoftFever/BambuStudio-SoftFever/discussions/450

[image]

Ixian, 2023-03-09:
Thanks for opening this. I'll add this isn't specific to the Sonic pad; any multi-printer Klipper/Moonraker/Mainsail/etc. setup will run in to it. 

Prusaslicer already allows for this over-ride (for this reason) so perhaps that could be a starting point?

SoftFever, 2023-03-17:
Implemented in 1.5.0