OrcaSlicer/OrcaSlicer #453: Print By Color

limited660, 2023-03-09:
**Is your feature request related to a problem? Please describe.**
When printing multiple colors and select print by object to reduce filament swaps, it would be nice to select print by color instead.

**Describe the solution you'd like**
A print by object type setting that instead prints by colors

**Additional context**
[image]

SwagDad99, 2023-03-11:
You could put all the black objects near each other, select them all, right-click, and choose assemble.  Then Softfever treats them as one object, and then can all huddle together in a corner...  Now, it is the same as print by color...

rkarlsba, 2023-04-06:
https://github.com/SoftFever/OrcaSlicer/issues/390

rkarlsba, 2023-11-26:
Can you please stop closing bugs as completed when they are just ignored? Better use a wontfix or cantbebothered or something. This one isn't completed.

SwagDad99, 2023-11-26:
Thank you. I appreciate it.  I feel the same way.  If it is just not addressed, it is still there...

From: Roy Sigurd Karlsbakk 
Sent: Saturday, November 25, 2023 5:00 PM
To: SoftFever/OrcaSlicer 
Cc: SwagDad99 ; Comment 
Subject: Re: [SoftFever/OrcaSlicer] Print By Color (Issue #453)

Can you please stop closing bugs as completed when they are just ignored? Better use a wontfix or cantbebothered or something. This one isn't completed.