OrcaSlicer/OrcaSlicer #460: Gcode variable for auxiliary part cooling fan

Ibot-11, 2023-03-10:
It would be nice to have a variable for the auxiliary part cooling fan which you can use in your startgcode.
Another idea is to use 0-255 for Marlin and 0-1 for Klipper for the min -> max speed, depending on your gcode flavor.

This would be perfect for controlling stuff like exhaust fans, nevermore or other fans.
My current workaround depends on the bed temperature:
```
{if bed_temperature_initial_layer_single > 80}
SET_FAN_SPEED FAN=nevermore_fan SPEED=1
SET_FAN_SPEED FAN=exhaust_fan SPEED=0
{else}
```

Maybe there is already a variable for this and I’m just unable to find it. If so please let me know how it's called.

SoftFever, 2023-03-11:
The auxiliary fan is controlled by “M106 P2” command.
We can override M106 in Klipper config to use the feature

Ibot-11, 2023-03-13:
Yes, that's right. I just thought it would be a useful addition since there is also a variable for the chamber temp already. Also then you would be able to use this variable in an if statement for example, which can be handy for marlin since there are no macros.

SoftFever, 2023-03-14:
I'm unsure if exposing fan speed as a variable is a good idea. I can't really think of a legit use case for that.

Anyway, Marlin support `M106 P<fan>` usage so this should work out of the box for Marlin machines.

midicdj1000, 2023-05-02:
Hi,

I have a simple way to parse the fan commands and change m106 p2 so it can turn on a klipper fan. It's nothing fancy I am new to this, but thought it could help others.

`[gcode_macro M106]
rename_existing: M106.0
gcode:
  {% if params.P is defined %}
    {% if params.P|int == 2 %}
      {% if params.S|int == 0%}
        SET_FAN_SPEED FAN=aux SPEED=0
      {% else %}
        SET_FAN_SPEED FAN=aux SPEED=1
      {% endif %}
    {% endif %}
  {% else %}
    { action_respond_info( "Setting fan speed to: {}%".format(((params.S|int|default(0))/255*100)|int) ) }
      M106.0 {rawparams}
  {% endif %}`

closed 2023-11-26 by github-actions[bot]