OrcaSlicer/OrcaSlicer #463: Enable Multimaterial load and unload time

abstractFlo, 2023-03-10:
**Is your feature request related to a problem? Please describe.**
I've been using an ERCF MMU with my Voron 2.4. It would be cool if we can setup this time

**Describe the solution you'd like**
[image]
Remove machine_load_filament_time and machine_unload_filament_time from hide section

(Would to that inside a PR but after compile i don't have default printers and filaments inside the dialog boxes

Basdiani, 2023-03-14:
How do you perform Tip shaping in softfever?

abstractFlo, 2023-03-16:
@Basdiani 
I use the Klipper Macros from ERCF to perform a tip shaping ;) there are commands to let the ercf do the shaping not the slicer.

RickDnamps, 2023-03-21:
Would be nice to have Tip Shapping has prusa slicer !!! Not the best to control that from the ERCF software... The parameter depends of the Filament...

closed 2023-11-26 by github-actions[bot]