OrcaSlicer/OrcaSlicer #468: inner-outer-inner/infill wrong order?

zviratko, 2023-03-11:
Either I don't get the purpose of the inner-outer-inner/infill or it is not working right

With 3 walls and infill the order looks like this:

1) outermost wall
2) innermost wall
3) middle wall
4) infill

 I thought it was supposed to work like this (for 3 walls):
1) middle wall
2) outer wall
3) inner wall(s)
4) infill

This would preserve dimensions because infill or multiple walls aren't pushing anything out if inexact while making overhangs work better (by the outer wall also sticking to the middle wall).
I am trying to think of a reason this should work differently and I also believe if that's the case then it is named wrong, as it clearly should either be laying inner wall first or be named "outer-inner-middle-infill" or something like that. I wasn't able to find a description for what exactly this is supposed to do, though...

SoftFever, 2023-03-12:
The order of sandwich mode for 3-wall should be:
Inner
Outer
Middle

zviratko, 2023-03-12:
This is not the case, I am attaching a 3mf with a simple cube that shows the problem.
[infillbug.zip](https://github.com/SoftFever/OrcaSlicer/files/10951199/infillbug.zip)

SoftFever, 2023-03-12:
Are you using the 1.5 dev build?
There is a regression bug in the dev branch.

You should be able to see the correct behavior in 1.4.5

zviratko, 2023-03-12:
Ah, sorry. Yeah, I am running some recent action build (1.5.0 merge I think).

Is there a way to achieve the ordering I thought it was supposed to be?
middle wall
outer wall
inner wall(s)
infill

Or should I open a RFE for that or something? 
Thanks

SoftFever, 2023-03-13:
I failed to see any benefits to printing in middle-outer order, can you elaborate the use case?
It is basically a worse version of inner-outer order IMHO.

zviratko, 2023-03-13:
Well, the reason why outer->inner is better for mechanical parts is accuracy. Overextrusion or inaccuracy in inner walls or infill won't affect the outer wall if it's laid down first.
The reason why most people don't actually use it is because it's much worse for overhangs, as the material is only holding partly on the previous layer making it free to ooze down. Having a wall next to it first makes it much easier to stick to it as well (surface tension?)

My proposal with middle (2nd outermost)->outer->inner/infill would be a compromise - you are still affected by inaccuracy in 1 wall but not by multiple inner walls

(btw I actually fail to see any benefit to using the sandwich mode as it's supposed to work, is that described somewhere?)

SoftFever, 2023-03-13:
It is described in previous release notes.Sent from my iPhone

closed 2023-03-17 by SoftFever

Wulfsta, 2023-11-16:
@SoftFever I would also be interested in the proposed perimeter mode, as a compromise between minimizing the impact of inner layers on the outer surface (less error introduced to the outer perimeter from a single interior perimeter than the cumulative error from several), maximizing the probability of a successful steep overhang, and reducing the impact of travel moves on the outer surface (as I believe was the original intent of sandwich mode). Would be happy to open a feature request or discuss offline if this thread is not appropriate.

skuep, 2024-01-13:
Found this by accident since I was about to open a feature request.

I would really love the wall order proposed above:
1. 2nd to outside wall
2. Outside wall
3. Rest of wall and infill

I use outer perimeter first, because it gives very good print quality (this is actually a tip from Ellis print tuning guide). But overhangs are always somewhat troublesome. This could be solved with this feature as a good compromise.

Is there any incentive/resources available for implementing something like this?

skuep, 2024-01-15:
Yeah, it's weird. Does anyone with knowledge of the codebase know where/who the root of this feature is?

Wulfsta, 2024-01-15:
It’s nontrivial, I looked into it in the past but didn’t have time to wrap my head around it to make the change. See [here](https://github.com/SoftFever/OrcaSlicer/blob/16158322362a10e58d0e974cad711ad9bc16b66d/src/libslic3r/PerimeterGenerator.cpp#L1741) and [here](https://github.com/SoftFever/OrcaSlicer/blob/16158322362a10e58d0e974cad711ad9bc16b66d/src/libslic3r/PerimeterGenerator.cpp#L2235).

skuep, 2024-01-15:
I see.. thanks for finding the location. from the looks of it, the code already does separate/index the separate perimeter lines. Maybe it's just a matter of changing the order of them in the slicing "result".

I am a bit more scared of the tools and libraries required to build the thing. I have to look up if there is some guidance for building (on Linux in this case).

zviratko, 2024-01-15:
> I am a bit more scared of the tools and libraries required to build the thing. I have to look up if there is some guidance for building (on Linux in this case).
> 
> Edit: it's right there in the readme 😁

If you do a pull request it should build it for you, I think (not a dev :))
I would be very happy to test it.

Wulfsta, 2024-01-15:
@skuep happy to help if you attempt it and get blocked - I'm not familiar with this code, but might be able to lend a hand if anything comes up.

SoftFever, 2024-01-16:
> @SoftFever I would also be interested in the proposed perimeter mode, as a compromise between minimizing the impact of inner layers on the outer surface (less error introduced to the outer perimeter [...]

So, your assumption is that the push force can be accumulated and passed to the outer wall all the way from the innermost wall?
I have doubts but still an interesting idea.
I will see if I can make an PR build for you all to test by end of this week.

Wulfsta, 2024-01-16:
> So, your assumption is that the push force can be accumulated and passed to the outer wall all the way from the innermost wall? I have doubts but still an interesting idea. 

My thought is more so that an error introduced that yields additional plastic (some bulge in the perimeter) would be additive due to the deposited plastic being in the path of the next perimeter. The error wouldn’t quite be additive, but more like applying ReLU before a sum. I am interested in testing it out though, so thanks for looking into implementing.

skuep, 2024-01-16:
Exactly.. see here for examples:
https://ellis3dp.com/Print-Tuning-Guide/articles/troubleshooting/extrusion_patterns.html#external-perimeters-first

Thanks a lot for your work! 🙂

SoftFever, 2024-01-16:
> My thought is more so that an error introduced that yields additional plastic (some bulge in the perimeter) would be additive due to the deposited plastic being in the path of the next perimeter. [...]

I mean, that's exactly why the outer-inner order exists in all slicers, and it's the precise reason why Orca introduced the inner-outer-inner order in the first place. You need to lay down the outer wall before the second outermost wall to avoid such errors.

SoftFever, 2024-01-16:
> Exactly.. see here for examples: https://ellis3dp.com/Print-Tuning-Guide/articles/troubleshooting/extrusion_patterns.html#external-perimeters-first And https://ellis3dp.com/Print-Tuning-Guide/articles/troubleshooting/bulging.html

That's just the normal 'outer wall first' mode. I have a strong feeling that you might have misunderstood why 'outer wall first' works and why Orca introduced the 'sandwich mode' to improve upon that

zviratko, 2024-01-16:
But the sandwich mode results in both poor overhangs and poor dimensional accuracy, as you are pushing more plastic around an already inexact (bulging) perimeter, which will ever so slightly affect the outer perimeter, even if it has somewhat cooled down, also what does it do to the next layer when there's a bulge on the top near the edge? If you lay down the second outermost perimeter first, then the outer perimeter, then the 3rd and up (going inward), then you get almost perfect accuracy but maintain good overhang angles and whatever excess material there is ends up pushing against infill doing no harm.

Just try it for yourself.
For a concete example, try printing this https://www.printables.com/model/928-yet-another-fidget-infinity-cube-v2
This model feels best when heavy, 80% infill works but using more walls would be better because then you also get more strength (not that it matters in this example). But with large number of perimeters (say 10?) and either inner-outer or sandwich mode, it usually gets fused. It works the same way on anything the requires precision (herringbone gears fidget from emmet does the same thing for example and also needs some overhangs if you don't print it with PLA).

skuep, 2024-01-16:
> That's just the normal 'outer wall first' mode. I have a strong feeling that you might have misunderstood why 'outer wall first' works and why Orca introduced the 'sandwich mode' to improve upon that

Yes, it's the outer wall first mode. I understand very well how this works. However it has the downside of overhangs being quite troublesome. Thus, an "improvement" in my book would be to use the advantage of 'outer wall first' (at least to some degree) but still be able to print overhangs. Yes, the second-outer wall can still produce defects on the outer wall, but to a much lesser degree than the regular wall order, where a lot more internal walls are printed, before the outer wall is printed.

> why Orca introduced the 'sandwich mode' to improve upon that

I think we are all wondering here, what exactly the improvement of the 'sandwich mode' is over 'outer wall first'? In fact, that is the whole reason this issue exists :-)

SoftFever, 2024-01-16:
> But the sandwich mode results in both poor overhangs and poor dimensional accuracy

I have conducted extensive testing by myself, and have also received feedback from a wide range of users, both through public forums and private messages. e.g. 
https://www.reddit.com/r/BambuLab/comments/130epbk/comment/jhwdnna/
https://github.com/SoftFever/OrcaSlicer/issues/18#issuecomment-1357348956

> , as you are pushing more plastic around an already inexact (bulging) perimeter, which will ever so slightly affect the outer perimeter, even if it has somewhat cooled down, also what does it do to [...]

How? In sandwich mode, when you lay down the outer wall, its neighboring inner wall has not been laid yet, so there is no effect from the inner wall on the outer wall. As the following screenshot demonstrates.
[image]

> If you lay down the second outermost perimeter first, then the outer perimeter, then the 3rd and up (going inward), then you get almost perfect accuracy 

Again, this is not correct.  "Lay down the second outermost perimeter first, then the outer perimeter" is exactly the reason why surface quality suffers.

> Just try it for yourself. For a concete example, try printing this https://www.printables.com/model/928-yet-another-fidget-infinity-cube-v2 This model feels best when heavy, 80% infill works but [...]

"It usually gets fused' has nothing to do with dimensional accuracy. For such print-in-place models, you just need to adjust the travel distance threshold to enforce retraction, even for very short travels.

SoftFever, 2024-01-16:
> > That's just the normal 'outer wall first' mode. I have a strong feeling that you might have misunderstood why 'outer wall first' works and why Orca introduced the 'sandwich mode' to improve upon that
> 
> Yes, it's the outer wall first mode. I understand very well how this works. However it has the downside of overhangs being quite troublesome. Thus, an "improvement" in my book would be to use the advantage of 'outer wall first' (at least to some degree) but still be able to print overhangs. Yes, this 'second-outer wall first' can still produce defects on the outer wall, but to a much lesser degree than the regular wall order, where a lot more internal walls are printed, before the outer wall is printed.

Sandwich is not trying to solve the overhang issue for outer wall first.
I think I mentioned it in one of the release notes, it's like outer wall first, is meant for models that has small overhang angles.
For model with large overhang angle, you can try the [Precise-wall](https://github.com/SoftFever/OrcaSlicer/wiki/Precise-wall) instead

> 
> > why Orca introduced the 'sandwich mode' to improve upon that
> 
> I think we are all wondering here, what exactly the improvement of the 'sandwich mode' is over 'outer wall first'? In fact, that is the whole reason this issue exists :-)

I think I need find sometime to write a WIKI for this. Meanwhile here is a very short description of it:
quoted from this [release note](https://github.com/SoftFever/OrcaSlicer/releases/tag/v1.3.4)

```
Add a new printer order - sandwich mode:
This new order is similar to the outer-wall-first mode in achieving the best dimensional accuracy. This new approach however avoids printing outer walls right after a long travel, which may cause artifacts on the surface in many cases.
```

skuep, 2024-01-16:
To be honest, I don't have any issue whatsoever with outer walls after "long travel moves" (which is what the sandwich mode is supposed to solve) but that is besides the point, since apparently some people are.

> On a four perimeter model the order would be `inner -> second-most outer -> outer -> third-most outer`.

I could live with that! Isn't that the ultimate compromise? Reiterating:
1. inner first -> Avoid issues with the long travel moves producing visible artifacts
2. second-most outer -> Better overhang performance
3. outer -> Still nice performance (given the circumstances)
4. fill in all missing perimeters

@SoftFever: Would this be a nice addition to the sandwich mode? I think it won't (negatively) effect the current users but still benefit others. And the change to the current sandwich mode is merely to prioritize the second-most outer wall above the outer wall (if I am not mistaken).

Wulfsta, 2024-01-16:
> @SoftFever: Would this be a nice addition to the sandwich mode? I think it won't (negatively) effect the current users but still benefit others. 

This should be a new perimeter mode altogether if it gets implemented, sandwich mode already works and has been extensively tested to show the benefit. No reason to make a breaking change to a working feature.

skuep, 2024-01-16:
That's also fine..
In that case I think it would be best to have the order as written in the beginning 
1. Second most outer perimeter
2. Outer perimeter 
3. Remaining perimeters

And call this something like "outer wall first plus". The reasoning behind this is, that this accomplishes what it's supposed to do down to 3 perimeters. Whereas @Wulfsta s order would print inner, middle outer, which is suboptimal in terms of print quality. Better would be 2nd-outer, outer, inner.

zviratko, 2024-01-16:
> "It usually gets fused' has nothing to do with dimensional accuracy. For such print-in-place models, you just need to adjust the travel distance threshold to enforce retraction, even for very short travels.

I can get 0.05mm knob to turn on Maker's Muse accuracy spinner and I get completely freely spinning gears on gear bearing spinners once removed from build plate, and that is with the standard Inner/Outer wall order but with 2 walls only. Once I use more, then it starts getting fused because everything gets pushed outward. And I have yet to see any sort of "gap fill" or single line fill in between existing walls to not have a visible bulge upwards indicating there's more material than needed, this is going to accumulate as well. And no, it's not an issue with flow rate, if I were to tune it to not do that I'd end up with extremely weak prints. I don't have a scientific explanatino for that happening, just a vague idea, but I don't think it matters. I
Let's agree to disagree on that, hopefully someone implements this and we get to test it for ourselves, I'm pretty confident the outcome will be net positive :)

skuep, 2024-01-20:
I've had a go at it by trying to just modify the existing inner-outer-inner mode as a first step. See my branch here:
https://github.com/SoftFever/OrcaSlicer/compare/main...skuep:OrcaSlicer:inner-outer-inner-reverse?expand=1

There are two ##if sections, the first one works (but doesn't really change the wall order at all), but as soon as I flip the first loop, OrcaSlicer crashes with all kinds of memory access errors (segmentation fault, invalid pointer, etc). The debug output however looks fine to me.

I slice the Micron Calibration Cube (but it shouldn't matter). If you want to try, you need to use Arachne and set the wall order in the GUI to ``inner-outer-inner``.

Does anyone have a clue why this happens? Otherwise these debug lines look fine to me

Works:
````
New Layer: Layer ID 75
Layer ID 75, Outer index 0, inner index 1, second inner index 2, maximum internal perimeter 3 
Inside out loop: Mapped perimeter index 0 to array position 0
Inside out loop: Mapped perimeter index 1 to array position 1
Outside in loop: Mapped perimeter index 2 to array position 2
Outside in loop: Mapped perimeter index 3 to array position 3
Layer ID 75, Outer index 4, inner index 5, second inner index 6, maximum internal perimeter 7 
Inside out loop: Mapped perimeter index 4 to array position 0
Inside out loop: Mapped perimeter index 5 to array position 1
Outside in loop: Mapped perimeter index 6 to array position 2
Outside in loop: Mapped perimeter index 7 to array position 3
Layer ID 75, Outer index 8, inner index 9, second inner index 10, maximum internal perimeter 11 
Inside out loop: Mapped perimeter index 8 to array position 0
Inside out loop: Mapped perimeter index 9 to array position 1
Outside in loop: Mapped perimeter index 10 to array position 2
Outside in loop: Mapped perimeter index 11 to array position 3
New Layer: Layer ID 0
New Layer: Layer ID 4
Layer ID 4, Outer index 0, inner index 1, second inner index 2, maximum internal perimeter 3 
Inside out loop: Mapped perimeter index 0 to array position 0
Inside out loop: Mapped perimeter index 1 to array position 1
New Layer: Layer ID 131
Layer ID 131, Outer index 0, inner index 1, second inner index 2, maximum internal perimeter 3 
Inside out loop: Mapped perimeter index 0 to array position 0
Inside out loop: Mapped perimeter index 1 to array position 1
Outside in loop: Mapped perimeter index 2 to array position 2
Outside in loop: Mapped perimeter index 3 to array position 3
Layer ID 131, Outer index 4, inner index 5, second inner index 6, maximum internal perimeter 7 
Inside out loop: Mapped perimeter index 4 to array position 0
Inside out loop: Mapped perimeter index 5 to array position 1
Outside in loop: Mapped perimeter index 6 to array position 2
Outside in loop: Mapped perimeter index 7 to array position 3
....
New Layer: Layer ID 129
Layer ID 129, Outer index 0, inner index 1, second inner index 2, maximum internal perimeter 3 
Inside out loop: Mapped perimeter index 0 to array position 0
Inside out loop: Mapped perimeter index 1 to array position 1
Outside in loop: Mapped perimeter index 2 to array position 2
Outside in loop: Mapped perimeter index 3 to array position 3
Layer ID 129, Outer index 4, inner index 5, second inner index 6, maximum internal perimeter 7 
Inside out loop: Mapped perimeter index 4 to array position 0
Inside out loop: Mapped perimeter index 5 to array position 1
Outside in loop: Mapped perimeter index 6 to array position 2
Outside in loop: Mapped perimeter index 7 to array position 3
New Layer: Layer ID 111
Layer ID 111, Outer index 0, inner index 1, second inner index 2, maximum internal perimeter 3 
Inside out loop: Mapped perimeter index 0 to array position 0
Inside out loop: Mapped perimeter index 1 to array position 1
Outside in loop: Mapped perimeter index 2 to array position 2
Outside in loop: Mapped perimeter index 3 to array position 3
Layer ID 111, Outer index 4, inner index 5, second inner index 6, maximum internal perimeter 7 
Inside out loop: Mapped perimeter index 4 to array position 0
Inside out loop: Mapped perimeter index 5 to array position 1
Outside in loop: Mapped perimeter index 6 to array position 2
Outside in loop: Mapped perimeter index 7 to array position 3

````

Crash:
[... 28 lines cut ...]
New Layer: Layer ID 121
Layer ID 121, Outer index 0, inner index 1, second inner index 2, maximum internal perimeter 3 
Inside out loop: Mapped perimeter index 1 to array position 0
Inside out loop: Mapped perimeter index 0 to array position 1
Outside in loop: Mapped perimeter index 2 to array position 2
Outside in loop: Mapped perimeter index 3 to array position 3
Layer ID 121, Outer index 4, inner index 5, second inner index 6, maximum internal perimeter 7 
Inside out loop: Mapped perimeter index 5 to array position 0
Inside out loop: Mapped perimeter index 4 to array position 1
Outside in loop: Mapped perimeter index 6 to array position 2
Outside in loop: Mapped perimeter index 7 to array position 3
....
New Layer: Layer ID 16
Layer ID 16, Outer index 0, inner index 1, second inner index 3, maximum internal perimeter 4 
Inside out loop: Mapped perimeter index 1 to array position 0
Inside out loop: Mapped perimeter index 0 to array position 1
Outside in loop: Mapped perimeter index 3 to array position 2
Outside in loop: Mapped perimeter index 4 to array position 3
Layer ID 16, Outer index 5, inner index 6, second inner index 7, maximum internal perimeter 8 
Inside out loop: Mapped perimeter index 6 to array position 0
Inside out loop: Mapped perimeter index 5 to array position 1
Outside in loop: Mapped perimeter index 7 to array position 2
Outside in loop: Mapped perimeter index 8 to array position 3
Segmentation fault (core dumped)
````

skuep, 2024-01-21:
Some progress: I successfully implemented this feature for the classic slicer (Arachne still would crash with my changes).
I might test print some parts today..
[image]

skuep, 2024-01-21:
This is my result. I compared ``inner-outer`` (regular), ``inner-outer-inner`` (sandwich), ``innerouter-outer-inner`` (new mode).

First picture shows the wall with **maximal unfavorable lighting**. This is very exaggerated and only visible, when the light comes straight from the top. Wall quality is best with ``inner-outer-inner`` (sandwich mode), because outer perimeter is printed without any disturbing neighbouring perimeter. This is to be expected .``inner-outer`` has the worst wall quality (this is in-line with what Ellis reports in his tuning guide above). ``innerouter-outer-inner`` (new mode) shows (in my opinion) a good compromise between the two. It's not as good as sandwich mode (or ``outer-inner`` for that matter), but way better than ``inner-outer``.

Second image shows overhang behaviour. (Un)fortunately the part does not have any problematic overhangs, but the highlighted region shows **some** effect. ``inner-outer`` has best overhang behaviour, and sandwich mode (and ``outer-inner`` for that matter) show worst overhang (as is to be expected, since outer perimeter has no neighbouring wall to "grab"). ``innerouter-outer-inner`` (new mode) does not have this problem. Looks almost identical to ``inner-outer``.

My verdict is: I would love to switch from ``inner-outer`` to the proposed new wall ordering as my *default*, since it does not have any disadvantage, but surface quality is better. For parts that require exceptional surface quality, I switch to sandwich or ``outer-inner`` mode (as I do already).

[image]
[image]

zviratko, 2024-01-21:
This is great, especially if it could become the default for inner-outer, as most people will just use it (especially now that 3d printing has become accessible to those who have no idea about anything :)). I don't see any downside to it, assuming it doesn't break anything unintentionally...

I am looking forward to testing this if someone makes a macOS build (but I could deal with Windows build as well I think). I have some models that really show the difference like the infinity cube. But I guess the best test would be simply this https://www.printables.com/model/693345-printer-test

skuep, 2024-01-23:
There you go. 
https://github.com/SoftFever/OrcaSlicer/pull/3809

Any idea, how to have the CI/CD system build the binary for you?

zviratko, 2024-01-23:
Hopefully, Github Actions will compile it, it just needs the maintainers to approve it
https://github.com/SoftFever/OrcaSlicer/actions

skuep, 2024-01-24:
> Hopefully, Github Actions will compile it, it just needs the maintainers to approve it https://github.com/SoftFever/OrcaSlicer/actions

The pull request auto build was approved. I think you should be able to start testing now 👍

zviratko, 2024-01-24:
Posted some tests in https://github.com/SoftFever/OrcaSlicer/pull/3809, someone should reproduce and we will probably move conversation there?