OrcaSlicer/OrcaSlicer #469: Can not open on Mac.  Says the file is damaged.  Have tried all versions. on a Mac Studio running Ventura

umpmanjoe, 2023-03-12:
**Describe the bug**
A clear and concise description of what the bug is.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

thewildmage, 2023-03-12:
This looks like a duplicate of #427. There's a [comment on that issue](https://github.com/SoftFever/OrcaSlicer/issues/427#issuecomment-1463891503) with a workaround that I think will help here.

SoftFever, 2023-03-15:
duplicated

closed 2023-03-15 by SoftFever