OrcaSlicer/OrcaSlicer #472: Split bridge speed to bridges and internal bridges (and cooling too)

Chodec80, 2023-03-12:
Hello everyone,
It would be nice to have individual speed setting for proper bridges and for internal bridges (first layer above infill). It is annoying to close an infill at the speed 40-50mm/s.

Related to the request above, would be also fine to separate cooling setting for bridges and internal bridges. 

Attached pictures are from Superslicer

Speed setting
[image]

Cooling setting
[image]

Thank you for consideration of this proposal

ab3dgeek, 2023-03-12:
That is a great idea! It could save a ton of time on some my prints too.

Chodec80, 2023-03-12:
> That is a great idea! It could save a ton of time on some my prints too.

yes, it can. Second benefit is "silence" because closing of the infill doesn't need full speed of the fan.

movinimage, 2023-09-03:
YES!  For a large flat object made of ABS/ASA or other high-warp materials, this is critical to avoiding failure. As it stands, I turn down the blower by hand while I sit by the printer! Otherwise the object pops of the build plate without fail.

Here is an example of something that fails every time unless blower is turned down...the object is 1/4" thick at the top of the first giant layer. (this object is about 7" wide.

[image]

Chodec80, 2023-11-23:
> I'm using 1.8.0 and it offers separate speed settings? Internal seems to default to 1.5 times faster.
> 
> [image]
Yes, they put it there few version back. For example, I use for internal bridges same speed as is defined for internal perimeter.

jonaseymour, 2023-12-12:
Yeah, this would be great. I recently had issues with layer adhesion with my PETG prints and narrowed it down to my internal bridge cooling being too high, but it was that high to get good external bridges.

Control over these would be great.

closed 2024-03-19 by github-actions[bot]