OrcaSlicer/OrcaSlicer #480: Klipper Gcode gets a M204 1000 emitted after a SET_VELOCITY_LIMIT ACCEL_TO_DECEL

Prutsium, 2023-03-13:
Latest nightly: https://github.com/SoftFever/OrcaSlicer/commit/5a986c18a1e811e88da1f61fedd9e88c78751b04
Sliced for Klipper

The Gcode get some wrong parameters omitted see example:
;AFTER_LAYER_CHANGE
;0.24
SET_VELOCITY_LIMIT ACCEL_TO_DECEL=500M204 S1000
SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY=10

As you see there is a wierd M204 standing there.
Attached Gcode (zipped) + 3mf (zipped)
[Rabbit Spring 2_0.2mm_ABS_3h52m.zip](
[Rabbit Spring 2.zip](https://github.com/SoftFever/OrcaSlicer/files/10961978/Rabbit.Spring.2.zip)
https://github.com/SoftFever/OrcaSlicer/files/10961975/Rabbit.Spring.2_0.2mm_ABS_3h52m.zip)

Prutsium, 2023-03-13:
Note .. switched back to 1.4.5 and all back to normal

SoftFever, 2023-03-14:
Just fixed.

Prutsium, 2023-03-14:
> Just fixed.

Thanks and great work!

closed 2023-03-17 by SoftFever