OrcaSlicer/OrcaSlicer #483: OrcaSlicer - layer height ?

Manfredo1952, 2023-03-15:
Should I report issues re the new version already, or just wait for official release I have compiled this myself

[image]

**Describe the bug**
set layer height 0.3 does not work (see screenshot)

 

**Expected behavior**
it does work in BambuSlicer 

**Screenshots**
 
[image]

**Printer model**

Bambu X1C

**Desktop (please complete the following information):**
 - OS: OSX  
 - Version 11.7.3

hliebscher, 2023-03-15:
it is not a bug see at #428

pls change in hotend settings from printer first

Manfredo1952, 2023-03-15:
Thanks. Overlooked this.