OrcaSlicer/OrcaSlicer #484: How can I print this shell with open hole

dojo90, 2023-03-15:
I have the following shell (https://anonfiles.com/g17110e8za/shell_stl)

[image]

I have already touched all the slicer settings, but every time, the hole is closed. Can someone please help me and tell me where to set the magic hook?

Thank you very much!

Pmant, 2023-03-16:
This is most likely a problem with your STL.
[image]

dojo90, 2023-03-17:
Thank you for your answers.
The problem is not with the STL file. When I repair the file, the slicer closes the hole with a lid. Just like the "bridge".

I also have the same problem when importing the *.STEP file.

I have also used some other models and everywhere I have the same phenomenon.

SoftFever, 2023-03-17:
Can you check with 1.5.0?
[image]
Looks fine.
If you still got issues with 1.5.0, please save it to 3mf project file and share it here.

closed 2023-11-26 by github-actions[bot]