OrcaSlicer/OrcaSlicer #496: Incorrect, strange icon

GP3D84, 2023-03-17:
**Describe the bug**
in list view the icon is not visible
this does not happen in other views

**To Reproduce**
by selecting list view

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS
 - Version Ventura (intel)
 - Latest OrcaSlicer (Release)

mmackowiak, 2023-04-18:
Still present in 1.6.1.

richard-scott, 2023-08-14:
I've just noticed this on 1.6.4-beta3.

richard-scott, 2023-08-24:
Still there in 1.6.6.

[image]

closed 2023-12-01 by github-actions[bot]