OrcaSlicer/OrcaSlicer #497: Bug in german translation (filament profile)

MasterTom99, 2023-03-17:
**Describe the bug**
In the filament settings the german translation for "first layer" is "Wände". The meaning of Wände is walls. It should be "erste Schicht" instead.

**3mf File for This Bug**

**To Reproduce** 
Change language to german. Start OrcaSlicer 1.5 and open the filament settings.

**Expected behavior**
"erste Schicht" should be used to translate "first layer".

**Screenshots**

**Printer model**
P1P

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 1.5

hliebscher, 2023-03-17:
@MasterTom99 da ich das so eingebaut habe will ich mal direkt antworten. Vorher hatte ich da erste Schicht stehen.
Das steht jetzt in der Überschrift das es um die erste Schicht geht. Das ganze steht auch nicht unter Filament sondern unter Geschwindigkeit. Da ja nicht der ganze erste Layer so gedruckt wird, dachte ich eine Trennung in Wände und Füllung und Bewegung trifft es auch wegen der Länge der Felder am besten. First Layer wäre für mich der ganze erste Layer und das ist es eben nicht.
Für Vorschläge bin ich offen rein übersetzt ist es natürlich falsch.

[image]

MasterTom99, 2023-03-18:
Hallo Heiko,

zunächst einmal vielen Dank für die Arbeit, die du in die Übersetzung gesteckt hast. Mein Englisch ist zwar ganz gut, bei den entsprechenden Fachbegriffen bevorzuge ich jedoch Deutsch.

Die Stelle, über die ich rede erreichst du, wenn du neben dem Filament auf den EDIT Knopf drückst:
[image]

Hier wird die Drucktemperatur des ersten Layers unabhängig von den anderen Layern eingestellt. Hier sollte "erste Schicht" statt "Wände" stehen, so wie es in 1.45 noch war.

Eine weitere Stelle, die nicht ganz konsistent ist ist das Druckmenü:
[image]

Der Punkt "Druckplatte" sagt noch nicht aus, was er bewirkt. Da du beim Exportieren die Druckplatte auch nur "Platte" nennst wäre mein Vorschlag folgender:

Aktuelle Platte drucken
Alle Platten drucken
Aktuelle Platte senden
Alle Platten senden
Exportiere aktuelle Platte als STL Datei
Exportiere alle Platten als STL Datei
G-Code exportieren.

Aber das ist wirklich Jammern auf hohem Niveau. Ich habe bisher keine abgeschnittenen Texte mehr gesehen, die nicht in die entsprechenden Fenster passten ;-)

Im allgemeinen fände ich es gut, wenn die Übersetzung so dicht wie möglich am Original bleibt. Sonst wird es schwierig, wenn man Tips aus englischen Foren umsetzen will und die entsprechenden Punkte nicht findet. Wenn du der Meinung bist, dass unter "Geschwindigkeit der ersten Schicht" besser "Wände" stehen muß, sollte man dies auch im englischen abändern.

Gruß
Thomas

hliebscher, 2023-03-19:
@MasterTom99 
Hallo Thomas
das sieht bei mir ganz anders aus ? schaue gleich mal nach... es ist natürlich nicht so leicht das anzupassen da nur einzelne Worte ohne Kontext übersetzt werden und ich habe nur MAC da sieht einiges anders aus und bestimmte Texte stehen da gar nicht... super wenn man Feedback bekommt so können wir das richtig gut machen...

closed 2023-03-19 by SoftFever

Th199403, 2023-05-14:
Hallo @hliebscher mit welchem Tool kann man die i18 MO Dateien bearbeiten? Und noch was  Wen ich ein Druck an den Drucker senden möchte steht dort aktualisieren statt senden.

commit a93be746 (2023-03-19): Merge pull request #522 from hliebscher/fix_de

fix #497 de
commit d6207042 (2023-03-19): fix #497 de