OrcaSlicer/OrcaSlicer #499: MacOS Catalina 10.15 support?

npike, 2023-03-17:
**Is your feature request related to a problem? Please describe.**
Bambu Studio 1.5.0 runs on MacOs Catlina 10.15, but OrcaSlicer does not.

**Describe the solution you'd like**
Would it be possible to compile OrcaSlicer to run on MacOs Catalina 10.15?

hliebscher, 2023-03-17:
i think it is not possible  can you update to 11.x
CMAKE_OSX_DEPLOYMENT_TARGET "11.3"

npike, 2023-03-17:
That's a bummer. My printer computer is a 2012 Mac Mini (latest update is 10.15) which runs prusaslicer and bambu studio fine.

Funkton, 2023-03-17:
[image]

A few versions back something was changed and support for Catalina was messed up. Should be a way to set the deploy target back.
Some of us are using computers which cannot be updated (work, toolchains, etc) and it would be nice to be able to use the enhanced features of this fork.

npike, 2023-03-18:
Bambu has theirs set to 10.15:

`set(CMAKE_OSX_DEPLOYMENT_TARGET "10.15" CACHE STRING "Minimum OS X`

hliebscher, 2023-03-18:
you can change it and make a test i have no old sdk for testing

Ryszard-wpl, 2023-03-18:
the worst thing is that since the disappearance of support for 10.15 I have not been able to compile SoftFever on 3 different computers. Yesterday I tried even on the Monterey mac with set(CMAKE_OSX_DEPLOYMENT_TARGET "10.15" CACHE STRING "Minimum OS X and later with original SopftFever set(CMAKE_OSX_DEPLOYMENT_TARGET "11.3" CACHE STRING "Minimum OS X

SoftFever, 2023-03-18:
"I can make another attempt to support 10.15, but I can't promise anything. Last time I saw errors that complained one of the features requires 11.3 onwards.

If there are any experienced Mac developers who can help look into this issue, we can probably address this faster.

paultanger, 2023-04-03:
Hi, does anyone know which older versions are still compatible that we could use? @Funkton do you recall?
Or could someone point me towards I could experiment with changing the min version parameter in a current version and build it and try it?

SoftFever, 2023-04-04:
@paultanger 
You can find `set(CMAKE_OSX_DEPLOYMENT_TARGET "11.3"` in these two files:
deps/CMakeLists.txt
CMakeLists.txt

Benderrodrigu3z, 2023-04-08:
dang I could really use this but same issue

Benderrodrigu3z, 2023-04-09:
I got it really far using Wine https://wiki.winehq.org/MacOS
Hit Escape to close Setup Wizard
Device tab not working but all other tabs work
[image]

Benderrodrigu3z, 2023-04-09:
[image]

Benderrodrigu3z, 2023-04-09:
clicking install driver does this and doesn't allow you to add the printers ip but looks like you could export to SD Card
[image]

Benderrodrigu3z, 2023-04-09:
note quiting terminal or that terminal window closes the program

npike, 2023-07-13:
Any further thoughts on this?

Benderrodrigu3z, 2023-11-17:
Fix itSincerely,Shane FSent from my iPhone13

SoftFever, 2023-12-30:
Fixed

closed 2023-12-30 by SoftFever