OrcaSlicer/OrcaSlicer #501: Preferences Window - Incorrect Translation for associate files to Orca Slicer in German

Launert, 2023-03-17:
**Describe the bug**
In the preference window, the subsection where you can select the filetypes to associate with Orca Slicer, in German translation the text still states to associate to Bambu Studio

**To Reproduce**
Select Language as German in setting windows.

**Expected behavior**
Files should be associated to Orca Slicer, not Bambu Studio. Description should reflect that fact.

**Screenshots**
English:
[image]

German: 
[image]

**Printer model**

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 21H2

hliebscher, 2023-03-17:
sorry is fixed now is only in windows version i use mac

Launert, 2023-03-17:
Perfect, no big deal anyways, just thought to report it.

hliebscher, 2023-03-18:
@Launert  i have chenge the translation again if you fine pls close the bug request
look at 
#515

Launert, 2023-03-19:
New translation look good. Thanks for the swift work!

closed 2023-03-19 by Launert