OrcaSlicer/OrcaSlicer #507: [Feature Request] Aux Fan Speed based on Bridge Length and/or Overhang Angle

Photogad, 2023-03-18:
**Is your feature request related to a problem? Please describe.**
I don't know if it's possible with Bambu Firmware but it irks me that there's no way I can find to adjust Aux Fan cooling speed based on bridge length or overhang angles. Being able to adjust this to 100% on the fly would allow for better bridging and overhangs. As it is now, it seems it can only be set to 1 constant speed (default of 70% for PLA)

**Describe the solution you'd like**
Settings to adjust Aux Fan Speed based on Bridge Length and/or Overhang Angle

**Describe alternatives you've considered**
none

**Additional context**
none

hliebscher, 2023-03-18:
it is in filament settings ...
you can set the speed for overhang and bridges and go slow down with print speed it works perfekt

mckellar11, 2023-03-20:
He said aux fan tho not part fan

Photogad, 2023-03-21:
> He said aux fan tho not part fan

Exactly I am talking about side Aux fan on X1

Photogad, 2023-05-14:
@SoftFever 
do you think this is possible to do in your fork or will it need to be addressed with BambuLab?

SoftFever, 2023-05-14:
@Photogad 
It seems the same thing as #843 
See my response there, which also applied to this.

Photogad, 2023-05-21:
> @Photogad It seems the same thing as #843 See my response there, which also applied to this.

@SoftFever thanks, I will comment there.

closed 2023-11-26 by github-actions[bot]