OrcaSlicer/OrcaSlicer #512: Is Arc Fitting activated by default? (It isnt active)

tommy536, 2023-03-18:
**Describe the bug**
A new sliced model with 1.5.0 shows up now G3 and G17 commands in the clipper console. Arc fitting is off. And by the way, the "c" shortcut do not work in preview mode.

**To Reproduce**
Upgraded from 1.4.5 and now these commands are present.

**Screenshots**
[image]

**Printer model**
Voron 2.4 (Klipper  v0.11.0-143-gc54d83c9)
Arc Welder off

**Desktop (please complete the following information):**
 - OS: Windows10

hliebscher, 2023-03-18:
for Mac M1 version it is fine no G3 code in file 

but for C command at first you to move the slider below  and it works only in a fresh sliced plate
if you change the plate in preview no g-code is shown (bug)

Hates1209, 2023-03-19:
I have also the same Unknow comands as G3 and G17. I use the same settings without any change of the config as before.

cotepat, 2023-03-19:
Same here on mac m1

SavouryMonkey, 2023-03-19:
> for Mac M1 version it is fine no G3 code in file
> 
> but for C command at first you to move the slider below and it works only in a fresh sliced plate if you change the plate in preview no g-code is shown (bug)

Nah, it's not. I'm on an M2 Pro, running the ARM version of it, and it generates arc command in the gcodes.

hliebscher, 2023-03-19:
the bug is the spiral lift. not the arc ...
G17
G3 Z1.3 I-.204 J-1.2 P1 F9000 ; spiral lift Z

SavouryMonkey, 2023-03-19:
> the bug is the spiral lift. not the arc ... G17 G3 Z1.3 I-.204 J-1.2 P1 F9000 ; spiral lift Z

Isn't G3 part of the whole Arc set? Like G2, G3, G17? So it shouldn't be in there in the first place, if disabled at least. 
Sliced the same part in 1.4.5 and 1.5 - no sign of arc commands in the old version - yet they are there in 1.5. 
Same settings overall.

hliebscher, 2023-03-19:
@cotepat have you set Z-Hop-Type to NORMAL ? for me its works

tommy536, 2023-03-19:
Yes, after switching to normal Z Hop no G3 and G17 is present. Why is spiral lift in 1.5 the default? I have never used this before.

SavouryMonkey, 2023-03-19:
> @cotepat have you set Z-Hop-Type to NORMAL ? for me its works

Yup. That fixed it.. totally overlooked. But why was it turned on in the first place? When never been used before I mean.

hliebscher, 2023-03-19:
i think it is default from BBS in 1.5.0 but klipper support arc. G2 G3

mckellar11, 2023-03-20:
I'm noticing slope z-hop but no documentation on that, what is it?

hliebscher, 2023-03-20:
@tommy536 
you can in printer cfg  enable arc commands G2 G3 G17

[gcode_arcs]
#resolution: 1.0

Ibot-11, 2023-03-22:
You have to add following macros. Somehow they are no longer included in the main README.md.
Is this intended @SoftFever ?

```
[gcode_arcs]
resolution: 0.1

[gcode_macro m201]
gcode:
  {% if 'X' in params or 'Y' in params %}
    {% set accel = (params.X|default(params.Y)|float,
                    params.Y|default(params.X)|float)|min %}
      SET_VELOCITY_LIMIT ACCEL={accel} ACCEL_TO_DECEL={accel * 0.5}
  {% else %}
    SET_VELOCITY_LIMIT
  {% endif %}

[gcode_macro m203]
gcode:
  {% if 'X' in params or 'Y' in params %}
    {% set speed = (params.X|default(params.Y)|float,
                       params.Y|default(params.X)|float)|min %}
    SET_VELOCITY_LIMIT VELOCITY={speed}
  {% else %}
    SET_VELOCITY_LIMIT
  {% endif %}

[gcode_macro M205]
gcode:
  {% if 'X' in params or 'Y' in params %}
    {% set corner_speed = (params.X|default(params.Y)|float,
                       params.Y|default(params.X)|float)|min %}
    SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY={corner_speed}
  {% else %}
    SET_VELOCITY_LIMIT
  {% endif %}

[gcode_macro M900]
gcode:
      SET_PRESSURE_ADVANCE ADVANCE={params.K}

# Make BambuStudio happy
[gcode_macro G17]
gcode:
```

closed 2023-11-26 by github-actions[bot]