OrcaSlicer/OrcaSlicer #517: How do I import all my filament profiles into the new Orca?

evilguy422, 2023-03-18:
I have a whole bunch of filament profiles set up in the previous version of SoftFever. Now Im using the Orca update by my profiles arent there. How can I import them?
Thanks

TheGoofy, 2023-03-18:
Same issue. Running Orcaslicer 1.5.0 on Windows 11
[image]

Manfredo1952, 2023-03-19:
> Same issue. Running Orcaslicer 1.5.0 on Windows 11 [image]

Have you been able to clear the issue as SoftFever suggested ?   (short while ago ?)

TheGoofy, 2023-03-19:
No. Actually I've seen @SoftFever's comment. Even though the files with the presets were already there, they don't show up in any of the presets in the application. I've also tried to use "File / Export / Export configs..." and "File / Import / Import configs...", but it didn't work. Note: custom process presets are also not there ... Maybe there is some Windows specific user-data-path, causing trouble due to application renaming.

SoftFever, 2023-03-19:
> 

Could you share one of the filament profiles here so I can investigate why it didn't load?

TheGoofy, 2023-03-19:
[Taulman Nylon PA.zip](https://github.com/SoftFever/OrcaSlicer/files/11011128/Taulman.Nylon.PA.zip)

Actually the log-file reveals the potential issue:

Taulman Nylon PA.json got a generic exception, reason = Invalid value provided for parameter filament_start_gcode: "bed adhesion tip: print on "hot-plate" with hairspray\n; filament start gcode\n{if (bed_temperature[current_extruder] >60)||(bed_temperature_initial_layer[current_extruder] >60)}M106 P3 S255\n{elsif (bed_temperature[current_extruder] >45)||(bed_temperature_initial_layer[current_extruder] >45)}M106 P3 S180\n{endif}"

TheGoofy, 2023-03-19:
(

> [Taulman Nylon PA.zip](https://github.com/SoftFever/OrcaSlicer/files/11011128/Taulman.Nylon.PA.zip)
> 
> Maybe the log-file reveals the potential issue:
> 
> Taulman Nylon PA.json got a generic exception, reason = Invalid value provided for parameter filament_start_gcode: "bed adhesion tip: print on "hot-plate" with hairspray\n; filament start gcode\n{if (bed_temperature[current_extruder] >60)||(bed_temperature_initial_layer[current_extruder] >60)}M106 P3 S255\n{elsif (bed_temperature[current_extruder] >45)||(bed_temperature_initial_layer[current_extruder] >45)}M106 P3 S180\n{endif}"

It's probably a red herring - the logs from the prior "bambustudio-softfever" version contains the sane error trace.

TheGoofy, 2023-03-19:
This is more interesting:
[image]

TheGoofy, 2023-03-19:
The log "create the main window" happens much later in bambustudio-softfever approx at line 200, and in orca it appears already at line 51 just after refusing to load the preset.

SoftFever, 2023-03-20:
@TheGoofy Thanks for the filament file.
Not clear why the `filament_start_gcode` causes parsing errors here. 
The code looks fine to me.
I will investigate it later. 
As a workaround, we can delete it.
Sorry for the hassles.

hotngui, 2023-06-25:
I am having the same issue. Running OrcaSlicer 1.6.3; I can see my filaments, which are just copied from the standard ones with a few minor changes, in the file system but they are not showing up in the UI such that I can select them.

zhp-pilot, 2023-06-26:
@hotngui go into Preferences (down arrow beside File menu) and turn on 'Auto sync user presets.' Then close and restart the app.

hotngui, 2023-06-26:
I tried that already and it does not work with v1.6.3 of the slicer. If I go back to v1.6.2 the syncing of my filaments works fine; but of course I lose all the other bug fixes that come with v1.6.3.

This is definitely a new bug in v1.6.3

closed 2023-11-26 by github-actions[bot]

DRIgnazGortngschirl, 2024-01-07:
@evilguy422 @TheGoofy @Manfredo1952 Please feel free to contribute your presets too https://github.com/DRIgnazGortngschirl/bambulab-studio-orca-slicer-presets, Thanks!