OrcaSlicer/OrcaSlicer #523: Start and End Area Bug is back

Ibot-11, 2023-03-19:
When using multiple buildplates the coordinates for the start and end area are wrong since it ignores that every plate does have its own origin point.
Therefore you can’t use adaptive mesh with multiple lates.
This bug already got patched but is back in version 1.5.0

hliebscher, 2023-03-19:
OS? Mac or WIN?

do you have a sample 3mf as zip file? for me ist works with MAC M1  V.1.5
i use the coordinates for the start and end area for bed mesh
Sample 3. Plate M557 X118.7:161.3 Y108.7:151.3 S25. the g code is correct

SoftFever, 2023-03-19:
It does look like it regressed for multi plates.

SoftFever, 2023-03-19:
fixed

hliebscher, 2023-03-19:
@SoftFever 
Now  the bug is back again !!! it works before we have fixed tis in older version now i get wrong values for 

M557 X-213.388:-176.704 Y87.9964:108.58 S25
M557 X{first_layer_print_min[0]}:{first_layer_print_max[0]} Y{first_layer_print_min[1]}:{first_layer_print_max[1]}

closed 2023-04-05 by SoftFever