OrcaSlicer/OrcaSlicer #525: Unable to set brim width for temp tower

eddydas, 2023-03-19:
I want to set brim width for temp tower but the setting is not respected.

1. Calibrate -> Temperature
2. Brim Width -> 20mm
3. Slice

Expected: the tower has 20mm brim
Actual: the tower has 5mm brim

Sometimes we need a much wider brim because we know that the print won't stick strong enough for the material we're printing with (e.g. P1P printing polycarbonate on textured PEI sheet - when the tower cools down it shrinks and would pull in slightly to make the bottom surface + around 5mm around it unstick, so we need extra brim to keep it stay on the bed)

hliebscher, 2023-03-19:
i think it is not a bug it is hard codet in source

("brim_type", new ConfigOptionEnum<BrimType>(btOuterOnly));
("brim_width", new ConfigOptionFloat(5.0));
("brim_object_gap", new ConfigOptionFloat(0.0));

and for all Temp Tower the same

SoftFever, 2023-03-19:
You can change them in object setting
[image]

closed 2023-03-20 by SoftFever

skintigh, 2025-07-23:
Also, unsure if this is related, but each time I printed with a different sized brim in that setting, Orca slicer crashed after successfully sending the file to my Bambu X1C. But that could be a coincidence, I need to use Orca more to be sure.

SamLR, 2025-12-09:
This still seems to be a problem and makes it tricky to print large brims for filaments like PHA which are very prone to curling on the first layer