OrcaSlicer/OrcaSlicer #527: Keyboard shortcuts for Prepare, Preview, Device, Project tabs

Stooovie, 2023-03-19:
**Is your feature request related to a problem? Please describe.**
Please add keyboard shortcuts for the main UI tabs (Prepare, Preview, Device, Project tabs) as well as the Slice command. Using mouse to navigate is is tiresome and slow.

**Describe the solution you'd like**
PrusaSlicer has numbers (1,2,3...) for views and ctrl (or cmd)-numbers (cmd-1, cmd-2...) for UI panels. I like that solution.

jbjhjm, 2023-11-17:
Second that! In general it seems difficult to find out what shortcuts even exist in Orca. Seems there's no documented list available at all? Some of the PrusaSlicer Shortcuts work, others are not. -> https://help.prusa3d.com/article/keyboard-shortcuts_1764
Using TAB to switch between prepare and preview is great already.
But I cannot find a shortcut for executing slicing.
The top bar commands may be the most pressed buttons / actions and I think they should have a shortcut!

closed 2024-02-25 by github-actions[bot]

nevack, 2024-02-25:
@SoftFever Can you reopen it? Issue is definitely present :)

SoftFever, 2024-02-25:
@nevack @Stooovie 
Do you got a chance to try it in the latest release?
While the shortcuts are not exactly same as PS, it works too.

https://github.com/SoftFever/OrcaSlicer/assets/103989404/754223d6-faf5-4ae1-80ac-4ccb891c68bf

You can check what shortcuts are available here:
[image]
[image]

nevack, 2024-02-25:
> Do you got a chance to try it in the latest release?

Yep, I know about Cmd+num to change view, use them regularly.

But this issue is about topmost toolbar.

While you can change between Prepare and Preview with Tab key, Device tab cannot be accessed with keyboard. I think PrusaSlicer doesn't have Device tab at all, it's from Bambu one.

SoftFever, 2024-02-25:
Please see my video, you can use ctrl+tab to circle through all views

nevack, 2024-02-25:
> Please see my video, you can use ctrl+tab to circle through all views

Oh, sorry, now I see.

Unfortunately, Cmd+Tab is a system shortcut for app switching, so OrcaSlicer cannot handle it properly.

[image]

I think many shortcuts with Ctrl by default are mapped to Cmd ones.
In this particular case this is wrong.
Ctrl+Tab is conventional for changing tabs in window even on macOS.
At least this works like this with Chrome and Finder on a mac.

nevack, 2024-03-06:
@SoftFever ping you here, as I still think this issue was closed wrongfully.

jbjhjm, 2024-06-21:
@SoftFever sorry to interrupt, but I too think this issue was closed wrongfully.
Application tabs should have good accessibility by keyboard.
Same goes for Process Global/Objects switch. 
(having the objects list hidden in there is my single PITA coming from Prusa/SuperSlicer btw.)