OrcaSlicer/OrcaSlicer #533: Please Add Height Range Modifier

kgearhardt, 2023-03-19:
One feature I miss the most having switched from printing on my MK3S+ to the P1P is the lack of the height range modifier in Bambu Studio and OrcaSlicer.

Knowing nothing about the related code in PrusaSlicer or SuperSlicer, it would be great to see that code ported over to OrcaSlicer in a future release.

Cheers!

kgearhardt, 2023-03-20:
I 3D print tap handle medallions for our keg distribution business, and part of the print is pausing to insert magnets.  On the Prusa I change the layer height from 0.2mm to 0.1mm so that I can pause after the "perfect" depth to insert the magnet, and then resume with a layer that's just 0.1mm above the magnets.  Then I'll switch back to 0.2mm heights.

It's arguably a luxury as I just printed a bunch on the P1P sliced with OrcaSlicer at 0.2mm layer heights, and it worked just fine.  I guess PrusaSlicer/SuperSlicer just spoiled me because that's what I started with.

Hope that makes sense.

alienboyxp, 2023-03-22:
Will be really useful, for example to create some tests like retraction, if no added before on Calibration menu :)

SoftFever, 2023-03-29:
Thanks for the explain.
Added to the backlog.

Meanwihile, closing this one as it's duplicated with #405

closed 2023-03-29 by SoftFever