OrcaSlicer/OrcaSlicer #536: Using G3 & G17 without Arc activated

SavouryMonkey, 2023-03-19:
Unknown command:"G3"
Unknown command:"G17"

Hello, 

for some reason it generates GCODE files with arc commands, even if the arc option is turned off. That results in 
a) a weird sounding voron, and 
b) unnessesary errors which waste print time

It'd be realy great if you could fix this. :)

hliebscher, 2023-03-19:
the bug is the spiral lift.  not the arc ...
G17
G3 Z1.3 I-.204 J-1.2 P1  F9000 ; spiral lift Z

SavouryMonkey, 2023-03-19:
> the bug is the spiral lift. not the arc ... G17 G3 Z1.3 I-.204 J-1.2 P1 F9000 ; spiral lift Z
> 
> pls set Z Hop Type to Normal then it works

> the bug is the spiral lift. not the arc ... G17 G3 Z1.3 I-.204 J-1.2 P1 F9000 ; spiral lift Z

Isn't G3 part of the whole Arc set? Like G2, G3, G17? So it shouldn't be in there in the first place, if disabled at least. 
Sliced the same part in 1.4.5 and 1.5 - no sign of arc commands in the old version - yet they are there in 1.5. 
Same settings overall.

hliebscher, 2023-03-19:
@SavouryMonkey 
if you set z Hop Lift Type in Extruder Settings to NORMAL then NO G2 G3 G17 in G-Code

SavouryMonkey, 2023-03-19:
> @SavouryMonkey
> if you set z Hop Lift Type in Extruder Settings to NORMAL then NO G2 G3 G17 in G-Code

Yup. That fixed it. Need to make sure to not turn on printer related settings without notification. ^^

closed 2023-03-19 by SavouryMonkey

hliebscher, 2023-03-20:
@SavouryMonkey 
you can in printer cfg  enable arc commands G2 G3 G17

[gcode_arcs]
#resolution: 1.0

Henk72, 2025-03-17:
@hliebscher 
I wouldn't use G2/G3 with Klipper. It will deteriorate quality and potentially slow the printer down.

If a spiral lift results in G2/G3 when arc fitting is disabled, this is for sure a bug. With a workaround to disable the spiral z-hop