OrcaSlicer/OrcaSlicer #538: Slicing inconsistencies between versions

SavouryMonkey, 2023-03-19:
[image]

I took the same part, and slices it with exact the same setup (Material, Printer, etc).. By exact i mean 1:1
It shows 1.5 got slower then 1.4.5 - and also shows inconsities between the windows and arm version of 1.5 (look at the infill).. 

Maybe someone could have a look at it.

hliebscher, 2023-03-19:
you have new features in 1.5  ... i think the speed settings are different between 1.4.5  and  slow down at overhangs
pls send the 3mf files as zip or look at the g-code for differences.
overhang walls  is twice the time looks like speed reduce 50% 
ist this classic mode or arachne?

SavouryMonkey, 2023-03-19:
> you have new features in 1.5 ... i think the speed settings are different between 1.4.5 and slow down at overhangs
> pls send the 3mf files as zip or look at the g-code for differences.
> overhang walls is twice the time looks like speed reduce 50%
> ist this classic mode or arachne?

It's Arachne in all 3 versions. Also I made sure that i compared every single setting, in every possible tab. They're identical, as are the material profiles and the printer with its settings. I'll upload the files in the morning.

mckellar11, 2023-03-20:
Pretty sure this is precise walls, notice the most stark difference is inner wall time, which I think got changed a bit or is just generating differently now

SavouryMonkey, 2023-03-20:
Like you've whished, here's the STL i tried on my comparison. It originated from here:
https://www.thingiverse.com/thing:4548782/files

Also i made sure to not overlook anything, every setting is, at least on the frontend, the same. Even the experimental precise wall option is turned off in all 3 instances of the slicer. 
[Landroid_Wheel_spike_slip-on.stl.zip](https://github.com/SoftFever/OrcaSlicer/files/11014761/Landroid_Wheel_spike_slip-on.stl.zip)

hliebscher, 2023-03-20:
we need your 3mf as zip to compare the settings

SavouryMonkey, 2023-03-20:
> we need your 3mf as zip to compare the settings

There ya go. 
[Archiv.zip](https://github.com/SoftFever/OrcaSlicer/files/11015048/Archiv.zip)

hliebscher, 2023-03-20:
i found the difference :-)
in Printer z Hop Type is in 1.5 at Spiral it takes more time.  if off than the time at all is equal

[image]
[image]

SavouryMonkey, 2023-03-20:
[image]
[image]

There's no spiral turned on.. It's weird, i did the same again, now with the corrected value of 20mm/s. Still different between mac and win. Also, i compared the project_settings.config - no difference (well, on the 1.5 versions) apart from the machies name. Yet one is faster then the other. 

Wir könnten auch deutsch reden :D 

[Archiv 2.zip](https://github.com/SoftFever/OrcaSlicer/files/11015370/Archiv.2.zip)

hliebscher, 2023-03-20:
ok ;-) mein Vergleich was zwischen 1.4.5 und 1.5 lade mir gerade die 3mf  alleine der G code ist 2000 Zeilen länger
bin dabei  zwischen mac und win wird es immer ein paar unterschiede geben. die Zeit ist ja auch nur geschätzt !
wir können auch gerne mal telefonieren ;-)

hliebscher, 2023-03-20:
> ok ;-) mein Vergleich was zwischen 1.4.5 und 1.5 lade mir gerade die 3mf alleine der G code ist 2000 Zeilen länger bin dabei zwischen mac und win wird es immer ein paar unterschiede geben. die Zeit ist ja auch nur geschätzt ! wir können auch gerne mal telefonieren ;-)

Hab übrigens auch den Landroid aber leider größere Räder und suche das File dafür ist ein 1500

SavouryMonkey, 2023-03-20:
> 

https://www.thingiverse.com/search?q=Landroid+L&page=1&type=things&sort=relevant

Da sind einige für den L dabei. :D 

Es ist schon irgendwie verwirrend, wieso ist in der config die spirale, obwohl im frontend der z hop offensichtlich auf normal steht? Das ist doch ein Bug oder? 

Klar, wenn du mir noch irgendwie nen hinweis gibst wie wir unsere Kontaktdaten austauschen, ohne das sie für hinz und kunz ersichtlich sind. :D

hliebscher, 2023-03-20:
hliebscher@gmail.com lösche ich gleich wieder

closed 2023-11-26 by github-actions[bot]