OrcaSlicer/OrcaSlicer #540: Feature request: Print order of objects

Prutsium, 2023-03-20:
Would it be possible to get similar implementation in Orca Slicer (damn will never get used to that name :) ) like SuperSlicer / PrusaSlicer that you can sort what file would be printed first?
The idea behind is that i often put a small square of 0.2 high to ensure nozzle offset is ok before the actual print starts.

In PS/SS this was done by just moving the concerning object in the top of the list and that one would be printed first.
SuperSlicer screenshots:
[image]
[image]

ab3dgeek, 2023-03-20:
Yeah that would be great 👍🏻🤓

H4kunaM4tata, 2023-05-03:
I have requested this a few months ago, perhaps my explanation was broken.
Glad to see this being implemented.

closed 2023-11-26 by github-actions[bot]

mrThe, 2024-01-23:
Bump. Can we please reopen this? That may not only solve OP issue, but also speedup print when you are printing a lot of objects at a time, because user can optimize part placement in a way it will require much less travel moves.
cc @SoftFever

Felix14-v2, 2024-03-12:
> GitHub bot: This issue was closed because it has been inactive for 7 days since being marked as stale.

WTH?

The most stupid bot ever made. The issue is still relevant.

mrThe, 2024-03-12:
@Felix14-v2 While bot is stupid for sure, issue is actually resolved: https://github.com/SoftFever/OrcaSlicer/pull/3685

But you need a nightly build of orca.

Felix14-v2, 2024-03-25:
> issue is actually resolved: #3685 
Not resolved. I still can't reorder parts of the model when printing them by layer.

HakunMatat4, 2024-03-25:
> > issue is actually resolved: #3685 
> 
> Not resolved. I still can't reorder parts of the model when printing them by layer.

It's resolved. I've been using for some time now on 2.0.0 nightly releases, I can print on the order I want by dragging the models around.
You need to change to object view tho.

Felix14-v2, 2024-03-26:
> I can print on the order I want by dragging the models around.

Dragging a model in the list does nothing when printing by layer is selected.

HakunMatat4, 2024-03-26:
> > I can print on the order I want by dragging the models around.
> 
> Dragging a model in the list does nothing when printing by layer is selected.

Yes, it does and I use it everyday.
In fact, I opened similar ticket myself in the past to solve this and it has been working just fine for releases now. Noisefox worked on it 

I'm on Linux and printing by layer NOT object, I can alter the order it is being printed by dragging the parts in the list.

Felix14-v2, 2024-03-27:
> Yes, it does and I use it everyday.

At least for me it doesn't work. 2.0.0 beta
https://github.com/SoftFever/OrcaSlicer/assets/75726196/0c4a91c1-1977-4921-af31-95c4661f708d

SoftFever, 2024-03-28:
> > Yes, it does and I use it everyday.
> 
> At least for me it doesn't work. 2.0.0 beta
> 
>  2024-03-27.15-50-19.trim.mp4

You need to change the setting here:
[image]

Felix14-v2, 2024-05-10:
> You need to change the setting here:

Still doesn't work.

https://github.com/SoftFever/OrcaSlicer/assets/75726196/01780066-1b52-4d16-b1a8-3ff0feaee296

I suspect it only works for individual models, not for the model parts.

mrThe, 2024-05-10:
>  it only works for individual models, not for the model parts.

It is. Don't load models as a single model. Or split it using built in split feature. It's sure not very convenient, but it is how it is.

Felix14-v2, 2024-05-11:
I have to load models with parts because this buggy software doesn't support moving individual models upper to print them in floating position. Orca forces your models to lie on a plate. But I need to reorder floating parts. пятница, 10 мая 2024г., 17:06 +03:00 от mr.