OrcaSlicer/OrcaSlicer #542: Since 1.5 the printer config isnt saved correctly

tommy536, 2023-03-20:
**Describe the bug**
After using 1.5, at every new start of the program the printer settings are an old / default one. When i saved the changes new, restart the program, the olf values registered on closing the program.

**Printer model**
Voron 2.4

**Desktop (please complete the following information):**
 - OS: Windows
 - 10

magicsoccer, 2023-03-21:
Same problem here.

Even if you rename the old directory and remove the duplicate entries in the new directory, they still appear when you start OrcaSlicer again.

MitjaHenner, 2023-03-25:
Deleting the default user folder helped.

Windows 11

tommy536, 2023-03-26:
How exactly do I have to proceed to avoid the problem? I deleted the Windows/User/DefaultUser folder but the problem persists.

magicsoccer, 2023-03-29:
My solution:

I deleted:
C:\Users\usernaem\AppData\Local\BambuStudio-SoftFever\
C:\Users\username\AppData\Roaming\BambuStudio-SoftFever\

I edited:
C:\Users\Andreas\AppData\Roaming\OrcaSlicer\user\1234567890\

closed 2023-11-26 by github-actions[bot]