OrcaSlicer/OrcaSlicer #543: Implement "Device View - reload"

MoppelMat, 2023-03-20:
My Klipper Mainsail device view stops working sometimes and I need to reload the web page. 

Sadly the Device view in OrcaSlicer has no reload option. I tried F5 and everything, but cannot get it to work. 

I have to restart the slicer to get the view to work again.

[image]

SoftFever, 2023-04-05:
Done.
F5 is shortcut key in 1.60

closed 2023-04-05 by SoftFever

asychev, 2023-07-03:
Does not work on OSX

therick0996, 2023-09-28:
i've found changing the printer, and then changing it back, refreshes it

negativesupply, 2023-11-27:
I've been having this issue too, on OSX. F5 doesn't work. CMD+R would be a good option because it matches safari for reloading pages.

hugoalves6, 2024-03-28:
Hello, i just got this problem too.
[image]

SoftFever, 2024-03-28:
> Hello, i just got this problem too. [...]

You can use right click menu to reload the page

ekle, 2025-03-28:
With the "Host Type" "CrealityPrint" right click does not work on linux.

Any other ways to force a reload?