OrcaSlicer/OrcaSlicer #545: Fill Bed With Instances

munkford, 2023-03-20:
**Is your feature request related to a problem? Please describe.**
No, this is a feature of PrusaSlicer that I'd like to see ported to Orca

**Describe the solution you'd like**
The ability to right-click an object and choose "Fill Bed With Instances". This is implemented well in PrusaSlicer, and the slicer will calculate the maximum number of instances it can fit, and then create those instances and auto-arrange them on the bed.

**Describe alternatives you've considered**
Currently in Orca, you need to manually Clone the object, then use Arrange/Auto-Arrange to lay them out.

**Additional context**
This may tie in with #448 which requests similar Instance handling to PS, and would likely be best implemented together.

[image]

radaiko, 2023-03-24:
I need this feature as well. 

Printing small parts where about 1000 fits on the Bambu bed is hard when you can only clone 100 parts as maximum.

dvomang, 2023-07-01:
Can we please add this feature to orca? It is super convenient in a lot of cases!

closed 2023-11-26 by github-actions[bot]