OrcaSlicer/OrcaSlicer #546: Placeholder for temperature offset

MitjaHenner, 2023-03-20:
I'm switching from super slicer where I'm using this start G-Code (Klipper):
```
START_PRINT BED_TEMP=[first_layer_bed_temperature] 
EXTRUDER_TEMP={first_layer_temperature[initial_extruder]+extruder_temperature_offset[initial_extruder]}
```

I guess Orca uses the same placeholders as Bambu, so I checked their wiki:
https://wiki.bambulab.com/en/software/bambu-studio/placeholder-list

```
START_PRINT BED_TEMP={bed_temperature_initial_layer[initial_extruder]}
EXTRUDER_TEMP={nozzle_temperature_initial_layer[initial_extruder]}
```

This works, but without the offset, it's the wrong temperature for the first layer. (The warm up and prime line, not the actual print)

hliebscher, 2023-03-21:
there is no value for  extruder_temperature_offset in Orca use nozzle_temperature and  nozzle_temperature_initial_layer
this are the 2 values from filament settings

MitjaHenner, 2023-03-25:
Thanks, this works fine, I had an error in my start gcode.

closed 2023-03-25 by MitjaHenner