OrcaSlicer/OrcaSlicer #547: Only one wall on top surface Arachne issue

blonestar, 2023-03-20:
**Only one wall on top surface does not work correctly in Arachne mode**
Single wall does work for very top layer but not the rest of the top layers.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Just switching between Arachne and Classic wall mod with the single wall option turned on.

**Expected behavior**
I guess all top layers should have single wall in Arachne mod as well, as it has in Classic wall mode.

**Screenshots**
[image]
[image]

**Printer model**
Voron Trident 250

**Desktop:**
 - OS: up-to-date Windows 10

Photogad, 2023-03-20:
Workign as intended, top layer is top layer. There is only one top layer.

blonestar, 2023-03-20:
I disagree. Have you look into my both screenshots btw?

blonestar, 2023-03-20:
When I said top layer I meant to say top surfaces, as the option is called.

blonestar, 2023-03-20:
[image]

npoulos, 2023-03-25:
”Surfaces” is plural which means there can be many top surfaces no matter the height.

blonestar, 2023-04-06:
Just confirming the issue in the v1.6.1

[image]

closed 2023-11-26 by github-actions[bot]