OrcaSlicer/OrcaSlicer #549: Exclude Object

LoboThirteen, 2023-03-20:
The Exclude Object function no longer works correctly. I always get the following error message in the screenshot. The script is of course in the Orca folder, just like in Soft Fever. In Soft Fever, the feature works as it should.

**Printer model**
RatRig V-Core3

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: OrcaSlicer 1.5.0

[image]

SoftFever, 2023-04-01:
It looks like you have solved the issue.
Closing it now.

FYI, OrcaSlicer will support Klipper `exclude_object` natively soon.

closed 2023-04-01 by SoftFever

rkarlsba, 2023-05-18:
Any idea when "soon" will be?