OrcaSlicer/OrcaSlicer #555: Blank Setup Wizard window on first launch

roboduid, 2023-03-21:
**Describe the bug**
Upon launching the slicer for the first time I just get a blank screen on the setup wizard

[image]

**Desktop (please complete the following information):**
 - OS: Windows 11 Pro (Ghost Spectre Superlite version)
 - Version 22H2

SoftFever, 2023-03-21:
Please refer to the installation section in the README

closed 2023-03-21 by SoftFever

roboduid, 2023-03-22:
This isn't fixed. I installed what was recommended in the Readme and still get a blank screen

roboduid, 2023-03-22:
I can't even close it out and start a new instance without using task manager

TexZeTech, 2023-03-22:
I will spin up a VM and see if I can replicate this problem but I suspect it's missing something in your none standard windows os.

**Version**
OrcaSlicer V1.50

**Describe the bug**

**To Reproduce**
Installed on fresh VM with nothing else installed

1st thing that happened. I got an error box "VCRUNTIME140_1.dll was not found"
2nd thing "MSVCP140.dll was not found" 
3rd "VCRUNTIME140.dll was not found"

and Orcaslicer didn't even show up.

I then installed MicrosoftEdgeWebView2RuntimeInstaller -> this one failed as it  had detected a more recent version installed
and vcredist2019_x64.exe -> no error

I then try to run OrcaSlicer again and the window pops up .... I'm then greeted with a "The application cannot run normally because your OpenGL version is lower than 2.0" followed by OrcaSlicer closing -> installed guest addons for virtual machine.

I then got it to the blank screen. ~~I suspect that not having EDGE  installed is the problem. To test me hunch I went into Ghost toolbox and used option#  and installed option# 1 edge stable~~

I have to go do something when I'm back I will poke this some more. But like I said at the start I suspect it's missing something in your none standard windows os.

**Expected behavior**
Correct resolution.

**Screenshots**

**Other notes**

**Virtual Machine**
Operating System: 22H2 Windows 11 PRO + Superlite Se + Defender
Version: Build: 22621.1265
Processors: 4 × Vcores
Memory: 8 GiB of RAM