OrcaSlicer/OrcaSlicer #556: Tip Shapping for Multi-material as ERCF

RickDnamps, 2023-03-21:
It would be nice to have Tip shapping has Superslicer or Prusaslicer.

That help to get Nice tip and to have better results with the multiple filament change.

See ramming button on the screen shot

[image]

[image]

Prutsium, 2023-03-21:
Not the optimal way but slice one object with the settings you want look into the Gcode and find the Ramming / Tip Shaping Gcode part ...
Drop this in Change Filament Gcode under Settings of the printer.
[image]

foxmckraut, 2023-03-21:
you got a cutter whatfor you need a tip shaping when the cutted part is the same on every switch?

kgearhardt, 2023-03-22:
I don’t think tip shaping for the AMS matters since the filament is cut inside the toolhead on every filament swap.  That’s one thing I don’t miss at all after setting it all up for my mmu2s.

Prutsium, 2023-03-22:
> I don’t think tip shaping for the AMS matters since the filament is cut inside the toolhead on every filament swap? That’s one thing I don’t miss at all after setting it all up for my mmu2s.

Not all people are using Orca Slicer for Bambu's ;)

kgearhardt, 2023-03-22:
> > I don’t think tip shaping for the AMS matters since the filament is cut inside the toolhead on every filament swap? That’s one thing I don’t miss at all after setting it all up for my mmu2s.
> 
> Not all people are using Orca Slicer for Bambu's ;)

Dang it, sorry I didn’t think about that :(

RickDnamps, 2023-03-22:
> > > I don’t think tip shaping for the AMS matters since the filament is cut inside the toolhead on every filament swap? That’s one thing I don’t miss at all after setting it all up for my mmu2s.
> > 
> > 
> > Not all people are using Orca Slicer for Bambu's ;)
> 
> Dang it, sorry I didn’t think about that :(
[Skip to content](https://github.com/SoftFever/OrcaSlicer/issues/556#start-of-content)
[SoftFever](https://github.com/SoftFever)
/
[OrcaSlicer](https://github.com/SoftFever/OrcaSlicer)
Public
forked from https://github.com/bambulab/BambuStudio
[Code](https://github.com/SoftFever/OrcaSlicer)
Issues
222
[Pull requests](https://github.com/SoftFever/OrcaSlicer/pulls)
[Discussions](https://github.com/SoftFever/OrcaSlicer/discussions)
[Actions](https://github.com/SoftFever/OrcaSlicer/actions)
Projects
1
[Wiki](https://github.com/SoftFever/OrcaSlicer/wiki)
[Security](https://github.com/SoftFever/OrcaSlicer/security)
[Insights](https://github.com/SoftFever/OrcaSlicer/pulse)
[Jump to bottom](https://github.com/SoftFever/OrcaSlicer/issues/556#issue-comment-box)
Tip Shapping for Multi-material as ERCF #556
Open
[RickDnamps](https://github.com/RickDnamps) opened this issue yesterday · 5 comments
Open
[Tip Shapping for Multi-material as ERCF](https://github.com/SoftFever/OrcaSlicer/issues/556#top)
#556
[RickDnamps](https://github.com/RickDnamps) opened this issue yesterday · 5 comments
Comments
[RickDnamps](https://github.com/RickDnamps) commented [yesterday](https://github.com/SoftFever/OrcaSlicer/issues/556#issue-1633125962)
It would be nice to have Tip shapping has Superslicer or Prusaslicer.

That help to get Nice tip and to have better results with the multiple filament change.

See ramming button on the screen shot

2023-03-20_22-21-16

2023-03-20_22-22-54

[Prutsium](https://github.com/Prutsium) commented [yesterday](https://github.com/SoftFever/OrcaSlicer/issues/556#issuecomment-1477590050)
Not the optimal way but slice one object with the settings you want look into the Gcode and find the Ramming / Tip Shaping Gcode part ...
Drop this in Change Filament Gcode under Settings of the printer.
image

[kgearhardt](https://github.com/kgearhardt) commented [2 hours ago](https://github.com/SoftFever/OrcaSlicer/issues/556#issuecomment-1479488118) • 
I don’t think tip shaping for the AMS matters since the filament is cut inside the toolhead on every filament swap? That’s one thing I don’t miss at all after setting it all up for my mmu2s.

[Prutsium](https://github.com/Prutsium) commented [2 hours ago](https://github.com/SoftFever/OrcaSlicer/issues/556#issuecomment-1479525028)
I don’t think tip shaping for the AMS matters since the filament is cut inside the toolhead on every filament swap? That’s one thing I don’t miss at all after setting it all up for my mmu2s.

Not all people are using Orca Slicer for Bambu's ;)

[kgearhardt](https://github.com/kgearhardt) commented [1 hour ago](https://github.com/SoftFever/OrcaSlicer/issues/556#issuecomment-1479536060)
I don’t think tip shaping for the AMS matters since the filament is cut inside the toolhead on every filament swap? That’s one thing I don’t miss at all after setting it all up for my mmu2s.

Not all people are using Orca Slicer for Bambu's ;)

Dang it, sorry I didn’t think about that :(

[RickDnamps](https://github.com/RickDnamps) commented [now](https://github.com/SoftFever/OrcaSlicer/issues/556#issuecomment-1479713959)
Yup my title say it... use multimaterial with ercf... I switched from superslicer to Orca cause superslicer is less maintened now by the guy who started yhe project and there is not many people who want to take the job back so its still on the old prusaslicer version.

Add a quote, <Ctrl+Shift+.>
Add code, <Ctrl+e>
Add a link, <Ctrl+k>
Directly mention a user or team
Reference an issue, pull request, or discussion
Add saved reply
Slash commands
Leave a comment
Remember, contributions to this repository should follow our [GitHub Community Guidelines](https://docs.github.com/articles/github-community-guidelines).
Assignees
No one assigned
Labels
None yet
Projects
None yet
Milestone
No milestone
Development
No branches or pull requests

Notifications
Customize
You’re receiving notifications because you authored the thread.
3 participants
@kgearhardt
@Prutsium
@RickDnamps
Footer
© 2023 GitHub, Inc.
Footer navigation
[Terms](https://docs.github.com/site-policy/github-terms/github-terms-of-service)
[Privacy](https://docs.github.com/site-policy/privacy-policies/github-privacy-statement)
[Security](https://github.com/security)
[Status](https://www.githubstatus.com/)
[Docs](https://docs.github.com/)
[Contact GitHub](https://support.github.com/?tags=dotcom-footer)
[Pricing](https://github.com/pricing)
[API](https://docs.github.com/)
[Training](https://services.github.com/)
[Blog](https://github.blog/)
[About](https://github.com/about)
RickDnamps commented Yup my title say it... use multimaterial with ercf... I switched from superslicer to Orca cause superslicer is less maintened now by the guy who started yhe project and there is not many people who want to take the job back so its still on the old prusaslicer version.

RefractingPrism, 2023-03-23:
> > > > I don’t think tip shaping for the AMS matters since the filament is cut inside the toolhead on every filament swap? That’s one thing I don’t miss at all after setting it all up for my mmu2s.
> > > 
> > > 
> > > Not all people are using Orca Slicer for Bambu's ;)
> > 
> > 
> > Dang it, sorry I didn’t think about that :(
> > [Skip to content](https://github.com/SoftFever/OrcaSlicer/issues/556#start-of-content)
> > [SoftFever](https://github.com/SoftFever
> 
> Yup my title say it... I use multimaterial with ercf... I switched from superslicer to Orca cause superslicer is less maintened now by the guy who started the project and there is not many people who want to take the job back so its still on the old prusaslicer version.

Use the happy hare software for ercf and i believe you can set it up to use the tip tuning through klipper so it’s not necessary to tune via slicer.

RickDnamps, 2023-03-23:
> Use the happy hare software for ercf and i believe you can set it up to use the tip tuning through klipper so it’s not necessary to tune via slicer.

Euh!! If you use many material like ABS, PLA, PETG, each filament need is own parameter .... in happy hare it's just one and you don't have all the parameter we have in the prusaslicer.... Normaly all those thing should be in the code of Bambulabs they just hide it cause they don't need it with thei printer...

closed 2023-11-26 by github-actions[bot]