OrcaSlicer/OrcaSlicer #558: Improver pressure advance line test

jezdinskyjiri, 2023-03-21:
Hi,
first of all thanks you for OrcaSlicer, it is awesome and I greatly appreciate calibration tool.

I would have one suggestions since I was calibrating some materials and I have issues with PETG that sticks too well and it is pain to remove.
One solution could be to use tower, however it is much easier to proper evaluate the line so I would prefer this.
Main issue is that there is so many strings and so many 0 and each of them needs to be peeled almost individually.

Could you make the test in a way that is all connected so in theory you could peel it as one piece?
Also could we get rid of extra 0s? 0.036 to be replaced with just 36? And of course the tutorial adjust it to reflect that? In worst case I would delete attest first zero, since it is completely redundant.

Thanks

Ibot-11, 2023-03-21:
Already requested this a while ago. #373 
I wouldn’t remove the zeros, since this can cause confusion especially with bowden setups where the PA value can go over 0.1
When they are all connected it’s no big deal if there are some more numbers or not.

NairoDorian, 2023-03-22:
Yes I would love this to be implemented as well, it is really hard sometimes with PETG and connecting every lines and every numbers would be the best solution.

closed 2023-11-26 by github-actions[bot]