OrcaSlicer/OrcaSlicer #560: crash on manjaro linux

yahbluez, 2023-03-21:
After running the appimage, 
the network binary gets loaded,
the loggin works, i see my avatar,
than the orca shows a message and terminates:

[image]

TexZeTech, 2023-03-21:
So I do get this error on & off when starting but I haven't logged in so I don't have Orcaslicer terminate on me.

[image]

After trying to duplicate this error I do indeed get a crash eventually. I also recall this happening in the previous release so I assumed that it wasn't a feature that was part of this fork. (much assumption was made here)

**Version**
OrcaSlicer V1.50 and older Appimage

**To Reproduce**
Start OcraSlicer -> Login -> Interface becomes non-responsive but doesn't show the classic unresponsive animation on the window (eg: greying out) then it closes to the desktop. Then open OcraSlicer again and it will result in a frozen interface again followed by a crash to the desktop.

**Desktop & hardware**:
Operating System: Manjaro Linux
KDE Plasma Version: 5.26.5
KDE Frameworks Version: 5.103.0
Qt Version: 5.15.8
Kernel Version: 5.15.102-1-MANJARO (64-bit)
Graphics Platform: X11
Processors: 8 × Intel® Core™ i7-3770K CPU @ 3.50GHz
Memory: 15.6 GiB of RAM
Graphics Processor: NVIDIA GeForce GTX 1080/PCIe/SSE2

**Alternative result in Ubuntu Vm**
On login, I'm prompted with a dialog to "Synchronize my personal data from Bambu Cloud"
if you press no -> It works as intended
if you press yes -> It works as intended

The JavaScript error about not finding a homepage also popped up but I think it's the symptom of another problem. 

**Dumb thing I tried**

I attempted to copy the config directory from the virtual machine to my main system running manjaro. Unfortunately, it worked as badly as when I attempted it the first time so that's a dead end on workaround at this point.

@yahbluez I also tried to login to BambuLabs Bambustudio in version 1.5.0.59 Appimage and it has effectively the same result on Manjaro. I also came across the error on their side that I believe is relevant.

[https://github.com/bambulab/BambuStudio/issues/760](url)

I think it will probably be best if this project gets away from .appimages as fast as possible or at least has build scripts for other systems besides ones that use dpkg only. 

I will also be the first to say I need to learn how to do that for myself. <.<

TexZeTech, 2023-03-22:
So using the [Aur package of bambustudio](https://aur.archlinux.org/packages/bambustudio) works with the network login so building from source seems to fix this problem. When I get a chance I'll play around with building OrcaSlicer from git and see if I can get a script going.

yahbluez, 2023-03-23:
@TexZeTech next time i will compile a version, but i like the concept of appimage and snap.

closed 2023-11-26 by github-actions[bot]