OrcaSlicer/OrcaSlicer #562: Option to pause print after flow calibration

hutje, 2023-03-21:
If you enable flow calibration in the settings of the printer before printing, it would be nice to have a gcode line with pause-command added right after so you can remove the lines before resuming the print.
When the flow calibration is turned off before starting the print, the pause command will not be executed.

Also nice when the object you want print is within the flow-calibration zone, the slicer prompts with a message to alert you can add a pause to be able to remove the flow-calibration lines.

rovili, 2023-03-21:
If you wanna go that advanced (you can add a pause manual in the start code of the printer), why don't you use the features of OrcaSlicer and calibrate the filament that you don't need to run the flow calibration?
Pressure Advance (flow calibration) is almost static for the same filament (same color) and the actual flow you need to calibrate anyway manually.

julie777, 2023-04-24:
Sadly, the gcode M0 to stop, put a message on the display, and wait for user input, is not available on Bambu Lab printers.

It is a very useful code for many reasons. I will post the request on the forum.

closed 2023-11-26 by github-actions[bot]