OrcaSlicer/OrcaSlicer #564: Still having an issue. Please don't close a report before it is actually solved.

roboduid, 2023-03-22:
**Describe the bug**
I'm still getting an issue where I have a constant blank screen when I launch the app. I can't even close the app without using task manager.
[image]

**Desktop (please complete the following information):**
 - OS: Ghost Spectre Windows 11 Superlite
 - Version 22H2

hliebscher, 2023-03-22:
i think the problem is your OS can you change the color ? or some packages are not in your lite  version

TexZeTech, 2023-03-22:
Well I'm dumb and put my answer in https://github.com/SoftFever/OrcaSlicer/issues/555#issuecomment-1479889834

Can someone merge the two I think its also solved but *shrug*

roboduid, 2023-03-22:
Actually this is fixed now. It was as simple as running as administrator

closed 2023-03-22 by roboduid