OrcaSlicer/OrcaSlicer #565: Connect to a klipper based printer directly from OrcaSlicer

esc247, 2023-03-22:
**Is your feature request related to a problem? Please describe.**
On Cura, you can install a plugin which enables you to send the print directly to the printer, as well as monitor the print from within the slicer. 

For me to use OrcaSlicer with other klipper based printers (VZBot) I have to export to an SD Card and then use the SD Card to print the sliced file.

**Describe the solution you'd like**
It would be great if I could connect my printer to the slicer directly and simply send the file directly to it over the network instead of having to us an sd card. The rest of the functionality (such as monitoring from within the slicer) is more of a nice to have.

esc247, 2023-03-23:
Thanks for the info, but for some reason I cannot see the WiFI icon in Orca, so I cannot add the IP. I ended up creating a full profile (edited from the Voron profile) for the VzBoT. I cannot find where to add the image of the VzBot 330 though...

esc247, 2023-03-23:
[VzBoT.zip](https://github.com/SoftFever/OrcaSlicer/files/11052812/VzBoT.zip)
I've added the config files that I based off of the demo printer from the Bambu Slicer wiki. When I added the demo printer to the config, I ran OrcaSlicer just to check if it loads, and it did. However, once I made the changes for my printer it wouldnt load. Cant figure out what I am doing wrong... :(

esc247, 2023-03-24:
I think I solved this. I'll add the latest version of the file for reference. Not sure if the VzBoT can be added to the list of supported printers for Orca (at least this should be a decent starting point) for others looking to add the VzBoT to the list :)

esc247, 2023-04-05:
Is there a way to do this so that I can be confident that all the references that I am replacing with my custom printer name are correct? I have a feeling that this issues is due to human error, where I am messing up the naming for the ref files... It can also be that it isn't loading as I am trying to set up a profile for a 0.6mm nozzle when the default is for 0.4mm?

TexZeTech, 2023-04-08:
I'm taking a look at this right now and see what I can do for ya.

esc247, 2023-04-09:
Great thanks!

esc247, 2023-04-13:
@TexZeTech any updates on this please? Thanks!

TexZeTech, 2023-04-14:
@esc247 I'm sorry about the slow reply I've been rather busy this week I did start looking at it but again haven't had the time to just crunch it out, so do not wait on me. When I get some free time I will continue.

TexZeTech, 2023-04-23:
@esc247 looks like there is a [beta release](https://github.com/SoftFever/OrcaSlicer/releases/download/v1.6.2-beta/OrcaSlicer_V1.6.2-beta-VzBot_Linux.zip) with it.

closed 2023-11-26 by github-actions[bot]