OrcaSlicer/OrcaSlicer #567: Not jumping to device tab after sending job

SwagDad99, 2023-03-22:
**Describe the bug**
The print dialog shows a counter, counting down until it jumps to the device tab.  It hits '0', and just goes back to the preview tab, instead of jumping to the device tab like it used to.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Send a job to the printer and wait

**Expected behavior**
It should jump to the device tab

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Printer model**
X1C

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 11 Home

Merlwynd, 2023-03-23:
Yes, same thing happens with me. Not a huge concern as I just open device tab…but it should go there automatically after countdown finishes.

SwagDad99, 2023-03-23:
I agree, it isn’t a big concern, just a minor bug that should be brought up as long as other fixes are going in...

There are actually times I am fine with it not jumping to the device tab.  Like when I am starting prints on multiple printers...

Don

From: Merlwynd 
Sent: Wednesday, March 22, 2023 11:40 PM
To: SoftFever/OrcaSlicer 
Cc: SwagDad99 ; Author 
Subject: Re: [SoftFever/OrcaSlicer] Not jumping to device tab after sending job (Issue #567)

Yes, same thing happens with me. Not a huge concern as I just open device tab…but it should go there automatically after countdown finishes.

Merlwynd, 2023-03-23:
I only have one printer, but I can see your point on multiple printers. If the majority of opinion is not to automatically jump to device page, I guess I'm ok with that as easy enough to click on device tab, but then they should remove the automatic message and the countdown clock...just want things to be consistent either way...

SwagDad99, 2023-03-24:
Exactly.  I agree with your statement.

Don

From: Merlwynd 
Sent: Thursday, March 23, 2023 12:06 PM
To: SoftFever/OrcaSlicer 
Cc: SwagDad99 ; Author 
Subject: Re: [SoftFever/OrcaSlicer] Not jumping to device tab after sending job (Issue #567)

I only have one printer, but I can see your point on multiple printers. If the majority of opinion is not to automatically jump to device page, I guess I'm ok with that as easy enough to click on device tab, but then they should remove the automatic message and the countdown clock...just want things to be consistent either way...

eboston, 2023-04-03:
> I only have one printer, but I can see your point on multiple printers. If the majority of opinion is not to automatically jump to device page, I guess I'm ok with that as easy enough to click on [...]

Before the latest release, it would switch to the device tab so it seems like something broke with the last release.  I can also see where not switching could be useful.  Instead of hardcoding an action, maybe an option can be added so the user can select whether to switch or not.

Marcel40625, 2023-04-06:
bug still exists with 1.6.0 for me!

julie777, 2023-04-09:
Strangely enough, although it doesn't jump to the device page, it does connect to the device and update status, and start the camera play. Both of these usually take noticeable time to start after sending a print so it is doing part of the switch task.

maziggy, 2023-04-10:
Same here with latest version.

Marcel40625, 2023-04-23:
Issue stills persists in 1.6.2-beta

julie777, 2023-04-24:
Please make this a priority. 
In addition to jumping to the device page, I think it would be a good idea to get rid of the 3 second countdown. Why not jump as soon as the job is sent.

There are two issues related to this. 
  1. because it doesn't jump immediately, I don't feel like waiting for the countdown and switch to another window
  2. Because I am not looking at the device page when the printer is parsing the gcode/3mf file, I don't notice that the parse failed and then later when I check the job status I find that it never actually started.

Marcel40625, 2023-05-01:
Issue stills persists in 1.6.2 !!!!

Merlwynd, 2023-11-17:
> GitHub bot: this issue is stale because it has been open for 90 days with no activity.

> GitHub bot: this issue is stale because it has been open for 90 days with no activity.

This should be closed as no longer an issue in latest versions….

closed 2024-02-25 by github-actions[bot]