OrcaSlicer/OrcaSlicer #568: Make speed mode a slider

SwagDad99, 2023-03-22:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**
Rather than have the four current choices of modes, (Silent, Standard, Sport and Ludicrous), a slider would be nice, being able to slide the mode positive, or negative. There are times I would like to print faster than Standard, but not as fast as Sport, or faster than Sport, but Ludicrous fails.  It would be nice to be able to 'slide' the speed up to %130.  Or down to %85 if necessary. 

Also, it would allow you to adjust the speed for certain parts/layers of a print.  Fast for lower detail layers, slower for higher detail layers, etc...

**Describe alternatives you've considered**
Sure, it is possible to adjust the speeds in the slicer before sending the print, but it would be nice to do it while printing.

**Additional context**
Add any other context or screenshots about the feature request here.

doudar, 2023-03-23:
This is something that would need to be implemented in the Bambu firmware. Until we have an open source firmware, you'll have to make this request at https://forum.bambulab.com/

SwagDad99, 2023-03-23:
OK.  Thank you

From: Anthony Doud 
Sent: Wednesday, March 22, 2023 7:19 PM
To: SoftFever/OrcaSlicer 
Cc: SwagDad99 ; Author 
Subject: Re: [SoftFever/OrcaSlicer] Make speed mode a slider (Issue #568)

This is something that would need to be implemented in the Bambu firmware. Until we have an open source firmware, you'll have to make this request at https://forum.bambulab.com/

closed 2024-02-19 by github-actions[bot]