OrcaSlicer/OrcaSlicer #572: Generic Linux Build Instructions with Pre-Requirement libs listed

A-nglea, 2023-03-23:
Great that ubuntu users can compile this. 

I use Fedora on cloning found it wouldn't build, despite changing apt for dnf, commenting out sections checking which gtk but still came up against library issues.
libgstreamerd-3 couldn't be found in Fedora's reps so I bailed.

The feature request is this: Ubuntu is a fine distro to start your Linux journey but many of us move on. Could your developers please recognise and address that and please provide for the four major Linux distros.

emelbardis, 2023-03-23:
Fedora user here. Agreed

nelsongraca, 2023-04-04:
I'm on the same boat but with Gentoo, it may be possible but knowing the devs don't seem to care doesn't make me very willing to make it work.

Even the AppImage that is supposed to be distro agnostic refuses to run.

TexZeTech, 2023-04-06:
The AppImage kinda sorta works on Manjaro and I did find an [AUR package](https://aur.archlinux.org/packages/orca-slicer) but I too agree.

nelsongraca, 2023-04-06:
@TexZeTech yep saw that one, and considered creating an ebuild for it, but the dependency on Wayland kills it for me as I'm on XFCE that has yet no Wayland support.

nelsongraca, 2023-06-19:
@chron0 No clue about wayland deps, I gave up on it and went to PrusaSlicer the devs don't seem to care much for proper Linux support other than Ubuntu, because of that I decided that I won't waste my time making an ebuild for it, would be different if there was developed interest.

closed 2023-11-20 by github-actions[bot]