OrcaSlicer/OrcaSlicer #575: Import Print Config from Gcode file

jhenley01, 2023-03-24:
Both PrusaSlicer and SuperSlicer have a feature that allows you to re-import all of the print settings from a previous gcode file. 

You can go to File > Import > Import Config and choose the gcode file. As long as the gcode file was generated from Prusaslicer or Superslicer it will import all of the print settings that were used. This is useful if you want to use the same settings that were used on a previous print but don't have the profile saved or the 3MF file.

It looks like OrcaSlicer is already outputting all the settings at the end of the the gcode file but there isn't a way to re-import them. If you go to File > Import > Import Config you aren't allowed to select a gcode file.

I've used this feature many times and it would be great to see it added to OrcaSlicer.

closed 2024-02-19 by github-actions[bot]

simpat1zq, 2024-03-15:
Is there any chance to get this feature on the roadmap?

foreachthing, 2024-04-05:
Just lost all my settings due to SSD-failure ... this would really be a life-saver right now 😢 
(_reminder to me to save all settings to "cloud" - only wish that `--data-dir` would work already_)

PJawZK, 2024-04-21:
The settings are in the g-code file at the bottom, like with previous renditions of this slicer, thinking of for instance PrusaSlicer.
So one can retrieve the info manually, but not automatically.

So the basis is there to be implemented, we just need to get SoftFever to find value in implementing it into OrcaSlicer.

www-dot-com, 2024-04-27:
It would be awesome if @SoftFever could implement this into Orca. I lost all my settings from Orca but I have them in g-code. It's a bummer that I can't just import them.

Aurasphere-au, 2024-09-28:
Was this ever implemented?

simpat1zq, 2024-09-28:
> Was this ever implemented?

No. So far it hasn't gotten any attention from any dev so it's unlikely to happen in the near future. I'm almost starting to think that some sort of script that passes the gcode and spits out a config file might be an option if someone is good with that type of stuff.